uunicorn/python-validity #123: openSUSE packaging

ArturGaspar, 2022-04-30:
I have created packages for openSUSE at https://build.opensuse.org/project/show/home:ArturGaspar:python-validity

Install instructions:

    zypper addrepo https://download.opensuse.org/repositories/home:ArturGaspar:python-validity/openSUSE_Tumbleweed/home:ArturGaspar:python-validity.repo
    zypper refresh
    zypper install fprintd-clients open-fprintd python-validity fprintd-pam

ArturGaspar, 2022-04-30:
Tested on a Lenovo X1 Carbon Gen 6 with `06cb:009a Synaptics, Inc. Metallica MIS Touch Fingerprint Reader`. Factory reset, enroll, verify and pam work.

xpufx, 2022-05-02:
@ArturGaspar Thanks for the package. It looks like I'm already using it via opi. This is a Lenovo T480S with the same fingerprint reader as yours. I too did the factory reset thing but enroll doesn't work. Any ideas?

> @t480s:~$ fprintd-enroll
> Impossible to enroll: GDBus.Error:net.reactivated.Fprint.Error.NoSuchDevice: No devices available

xpufx, 2022-05-02:
I only had the python-validity from @ArturGaspar repo. I installed the rest of the packages at the top of this issue and it worked after.

karoltheguy, 2024-06-03:
@xpufx I have the same laptop as you (T480) and have the same error msg as you after installing the packages from @ArturGaspar .
I know it has been a couple years but I am wondering if you got it working.

thanks

zabulonian, 2024-07-03:
On a Lenovo X1 Carbon Gen 6 and this unfortunately does not seem to work any longer. Reset the reader in bios, installed the repo & packages as per instructions, but when attempting to enroll: "Impossible to enroll: GDBus.Error:net.reactivated.Fprint.Error.NoSuchDevice: No devices available"

FWIW, fwupdmgr also is unable to see the fingerprint reader. I also noticed that these packages predate an update to the main repository which fixes certain directories not existing after reboot, and it may be affected by this bug? As I could not find the ./dbus_service/ directory. This could just be me overlooking/misunderstanding something, though

roelandjansen, 2024-12-22:
@zabulonian I just installed the whole stuff and I could enroll a fingerprint without your mentioned issue. 
Not saying I can log in with a fingerprint but at least ::

~/src/python-validity> fprintd-list roeland
found 1 devices
Device at /net/reactivated/Fprint/Device/0
Using device /net/reactivated/Fprint/Device/0
Fingerprints for user roeland on DBus driver (press):
 - #0: right-index-finger

did you check:

sudo systemctl status python3-validity
● python3-validity.service - python-validity driver dbus service
     Loaded: loaded (/usr/lib/systemd/system/python3-validity.service; enabled; preset: disabled)
     Active: active (running) since Sun 2024-12-22 19:28:58 CET; 5min ago

zabulonian, 2024-12-23:
@roelandjansen h
Hi, I'm glad it worked for you. I see you've cloned the source directory - that may be where I'm going wrong? I haven't done that, as I didn't realize I needed to. What did you do after you did so?

Otherwise, I just ran through the install instructions at the top, ran into the same GDBus error, and then followed the instructions here to try to clear it up: https://github.com/uunicorn/python-validity?tab=readme-ov-file#errors-on-startup

Unfortunately, the factory-reset.py script returns traceback errors. I'm unsure why, because as I was about to check for missing python libraries or something, I ran it again after waiting, and it seems to have worked. Still, after re-enabling the python3-validity service, I am unable to enroll due to, again, the GDBus error.

roelandjansen, 2024-12-24:
the latter, last line about unlocking is also what I faced.  something to look at. 

and yes I cloned/built/installed.