uunicorn/python-validity #132: Lenovo X1 Carbon Gen6 - Fedora

alexjfinch, 2022-06-30:
Hi all,

I have had this driver working on Fedora in the past with a few workarounds and I recently did a full reinstall and used the copr repo for the install - sadly it won't run.

Installs fine but then when running "fprind-enroll" the light on the sensor flickers and the command line is waiting for you to enrol a finger but it doesn't respond.

Also just tried it again and found that the service has an error which might help;

```
ERROR:root:App payload before secure connection established
Traceback (most recent call last):
File "/usr/lib/python-validity/dbus-service", line 157, in run
sensor.enroll(usr, index, update_cb)
File "/usr/lib/python3.10/site-packages/validitysensor/sensor.py", line 828, in enroll
self.create_enrollment()
File "/usr/lib/python3.10/site-packages/validitysensor/sensor.py", line 749, in create_enrollment
assert_status(tls.app(pack('<BL', 0x69, 1)))
File "/usr/lib/python3.10/site-packages/validitysensor/tls.py", line 152, in app
return self.parse_tls_response(self.usb.cmd(self.make_app_data(b)))
File "/usr/lib/python3.10/site-packages/validitysensor/tls.py", line 372, in make_app_data
raise Exception('App payload before secure connection established')
Exception: App payload before secure connection established
```
I'm a little bit green with linux and debugging stuff, so if someone can help point me in the right direction for finding out;

1.) The above error
2.) What the error message is when I try to enroll and the scanner itself activates and then instantly deactivates (might help figure out what's blocking the device from starting).

Cheers!

alexjfinch, 2022-06-30:
Ah ha! so I restarted the service, tried to enroll my finger and then checked the status of the service and found this error message;

```
Traceback (most recent call last):
File "/usr/lib/python3.10/site-packages/validitysensor/usb.py", line 128, in wait_int
resp = self.dev.read(131, 1024, timeout=100)
File "/usr/lib/python3.10/site-packages/usb/core.py", line 1029, in read
ret = fn(
File "/usr/lib/python3.10/site-packages/usb/backend/libusb1.py", line 864, in intr_read
return self.__read(self.lib.libusb_interrupt_transfer,
File "/usr/lib/python3.10/site-packages/usb/backend/libusb1.py", line 954, in __read
 _check(retval)
File "/usr/lib/python3.10/site-packages/usb/backend/libusb1.py", line 602, in _check
raise USBTimeoutError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBTimeoutError: [Errno 110] Operation timed out
                                           
During handling of the above exception, another exception occurred:
                                           
Traceback (most recent call last):
File "/usr/lib/python-validity/dbus-service", line 157, in run
sensor.enroll(usr, index, update_cb)
File "/usr/lib/python3.10/site-packages/validitysensor/sensor.py", line 844, in enroll
raise e
File "/usr/lib/python3.10/site-packages/validitysensor/sensor.py", line 832, in enroll
 self.capture(CaptureMode.ENROLL)
File "/usr/lib/python3.10/site-packages/validitysensor/sensor.py", line 711, in capture
b = usb.wait_int()
File "/usr/lib/python3.10/site-packages/validitysensor/usb.py", line 135, in wait_int
raise CancelledException()
  validitysensor.usb.CancelledException
```
I shall do a little more digging but if anyone understands the above, then great! :)

lckarssen, 2022-08-01:
Not sure if this will be of any direct help to you, but today I installed `python-validity` on my X1 Carbon Gen 6 with Ubuntu 22.04 following the installation instruction in the README. So far it works almost flawlessly (almost because I enrolled two fingers so far and it seems not to recognise my first finger. Not sure if this is a problem with my finger or the software :grin:. But that's not for this issue.)

Tkong1, 2022-11-04:
Hey @lckarssen i have the behavior that you describe, that my first enrolled finger print gets not recognized. How did you enroll a second finger?

alexjfinch, 2022-11-05:
I ran fprintd-enroll today and it worked perfectly. Following this the option to enable the fingerprint login didn't appear in settings for gnome, so I ran "sudo authselect enable-feature with-fingerprint" and rebooted. Authentication now works with fingerprint.

closed 2022-11-05 by alexjfinch

lckarssen, 2022-11-05:
> Hey @lckarssen i have the behavior that you describe, that my first enrolled finger print gets not recognized. How did you enroll a second finger?

@Tkong1 I enrolled other fingers via Ubuntu (Gnome's) user manager: Settings → User → Unlock → Click on "Enabled" next to "Fingerprint login" → click the + button.

Tkong1, 2022-11-05:
@lckarssen Thank! I also enrolled a second finger and it works. Maybe this is really a software issue.