uunicorn/python-validity #133: Proper set up guide for Manjaro Thinkpad T480

DominikBobinski, 2022-06-30:
Hi,
Thank you very much for creating python-validity. I've been using it for a couple of months now. Throughout this time i was creating an install guide for my future self which now is at the point where it might be helpful to others. It contains detailed steps and the end product is a fingerprint sensor that works flawlessly even after sleep or hibernation.

May I add my guide as a new "Manjaro" section in the readme file?

DominikBobinski, 2022-07-10:
I decided not to wait and made a pull request.
https://github.com/uunicorn/python-validity/pull/135
Please let us know if it works for you!

jmgant, 2023-02-14:
I have been wrestling with this on my Arch thinkpad. I was able to get shell login via either fprint/passwd. I could do a KDE screen unlock the same. Could not get SDDM login to work.

When I tried your method, SDDM prompted for a passwd, then rejected it, The fprint sensor then lit up, but it would not accept my input. Switching tty allowed me to fprint login via shell because I had pam_fprintd.so entry in system-local-login.

I had created a file in pam.d called common-auth. It includes these lines:
``
auth [success=2 default=ignore] pam_fprintd.so max_tries=1 timeout=10
auth [success=1 default=ignore] pam_unix.so nullok_secure try_first_pass
```
I then added an "include" line in the pam.d sddm file referencing this.

This is my first time doing anything with pam. What am I missing?