uunicorn/python-validity #136: Can't enroll on Thinkpad T470 with 138a:0097 Exception: Failed: 0404

lallenlowe, 2022-07-17:
- Thinkpad T470
- Fedora 36
- ID 138a:0097 Validity Sensors, Inc.
- installed python-validity using copr

When I run `fprintd-enroll` I get the following error: `Impossible to enroll: GDBus.Error:net.reactivated.Fprint.Error.NoSuchDevice:`

`sudo systemctl status python3-validity` shows:
```
× python3-validity.service - python-validity driver dbus service
     Loaded: loaded (/usr/lib/systemd/system/python3-validity.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Sat 2022-07-16 17:34:15 MDT; 36min ago
   Main PID: 2653 (code=exited, status=1/FAILURE)
        CPU: 171ms

Jul 16 17:34:15 fedora-thinkpad systemd[1]: python3-validity.service: Scheduled restart job, restart counter is at 5.
Jul 16 17:34:15 fedora-thinkpad systemd[1]: Stopped python3-validity.service - python-validity driver dbus service.
Jul 16 17:34:15 fedora-thinkpad systemd[1]: python3-validity.service: Start request repeated too quickly.
Jul 16 17:34:15 fedora-thinkpad systemd[1]: python3-validity.service: Failed with result 'exit-code'.
Jul 16 17:34:15 fedora-thinkpad systemd[1]: Failed to start python3-validity.service - python-validity driver dbus service.
```
`sudo validity-sensors-firmware` gives:
```
Downloading https://download.lenovo.com/pccbbs/mobiles/nz3gf07w.exe to extract 6_07f_lenovo_mis_qm.xpfwext
Extracting "version 5.3.3542.26 (NZ3GF07W)" - setup data version 5.5.7 (unicode)
 - "code$GetExtractPath$/WBF_Drivers/6_07f_lenovo_mis_qm.xpfwext"
Done.
Found firmware at /tmp/tmpfrcte_g4/code$GetExtractPath$/WBF_Drivers/6_07f_lenovo_mis_qm.xpfwext
```

`sudo python3 /usr/share/python-validity/playground/factory-reset.py -v` gives:
```
Traceback (most recent call last):
  File "/usr/share/python-validity/playground/factory-reset.py", line 6, in <module>
    factory_reset()
  File "/usr/lib/python3.10/site-packages/validitysensor/sensor.py", line 86, in factory_reset
    assert_status(usb.cmd(b'\x10' + b'\0' * 0x61))
  File "/usr/lib/python3.10/site-packages/validitysensor/util.py", line 12, in assert_status
    raise Exception('Failed: %04x' % s)
Exception: Failed: 0404
```
Please let me know what else I can try

IceCloud76, 2022-07-17:
I just ran into this issue with Fedora 36 on my T480 a few hours ago. Run `sudo systemctl stop python3-validity` then `sudo dnf remove open-fprintd fprintd-clients fprintd-clients-pam python3-validity`
After that, I installed all those packages again (`sudo dnf remove open-fprintd fprintd-clients fprintd-clients-pam python3-validity`)  and rebooted, then followed the guide like normal. Then it worked fine and I didn't get the 0404 error anymore

lallenlowe, 2022-07-17:
@IceCloud76 Dang, I was really hoping that would work for me. I see no change at all.

lallenlowe, 2022-07-18:
FYI running `sudo validity-led-dance` _does_ indeed cause the fingerprint reader LED to blink. I feel like I am _so_ close.

Sesam31, 2022-07-27:
Same problem here with fedora 36 but running the 9a sensor on a thinkpad X280.

Sesam31, 2022-07-27:
Nevermind I just followed #89 and the problem is fixed :)