uunicorn/python-validity #137: Listed as supported on fprntd page 04f3:0c28 Elan Fingerprint reader is not recognised by python-validity.

amikot, 2022-07-26:
Hi, I'm trying to get this thing working, but having trouble with it.
Device is listed on fprintd devices list, but is not working with python-validity?

Is there anything I'm doing wrong or it's a bug or my device is really not supported even if it's listed as supported?

I'm sure that device itself is Okay, because just tested it on Windows10. 

```
lsusb
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 004: ID 05c8:0374 Cheng Uei Precision Industry Co., Ltd (Foxlink) HP EliteBook integrated HD Webcam
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 007: ID 04f3:0c28 Elan Microelectronics Corp. fingerprint sensor [FeinTech FPS00200]
Bus 003 Device 002: ID 3938:1047 MOSART Semi. 2.4G Wireless Mouse
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 006 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
fprintd-enroll
list_devices failed: 
```

```
sudo systemctl status python3-validity
× python3-validity.service - python-validity driver dbus service
     Loaded: loaded (/lib/systemd/system/python3-validity.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2022-07-26 10:04:19 BST; 17s ago
    Process: 6199 ExecStart=/usr/lib/python-validity/dbus-service --debug (code=exited, status=1/FAILURE)
   Main PID: 6199 (code=exited, status=1/FAILURE)
        CPU: 587ms

Jul 26 10:04:19 a-laptop dbus-service[6199]:     init.open()
Jul 26 10:04:19 a-laptop dbus-service[6199]:   File "/usr/lib/python3/dist-packages/validitysensor/init.py", line 47, in open
Jul 26 10:04:19 a-laptop dbus-service[6199]:     usb.open()
Jul 26 10:04:19 a-laptop dbus-service[6199]:   File "/usr/lib/python3/dist-packages/validitysensor/usb.py", line 50, in open
Jul 26 10:04:19 a-laptop dbus-service[6199]:     self.open_dev(dev)
Jul 26 10:04:19 a-laptop dbus-service[6199]:   File "/usr/lib/python3/dist-packages/validitysensor/usb.py", line 62, in open_dev
Jul 26 10:04:19 a-laptop dbus-service[6199]:     raise Exception('No matching devices found')
Jul 26 10:04:19 a-laptop dbus-service[6199]: Exception: No matching devices found
Jul 26 10:04:19 a-laptop systemd[1]: python3-validity.service: Main process exited, code=exited, status=1/FAILURE
Jul 26 10:04:19 a-laptop systemd[1]: python3-validity.service: Failed with result 'exit-code'.

```
Tried to start the service that was showing errors:
`sudo systemctl start python3-validity`

But it didn't help. 

Then tried to perform factory reset of the device:
```
sudo systemctl stop python3-validity
sudo validity-sensors-firmware
Traceback (most recent call last):
  File "/usr/bin/validity-sensors-firmware", line 92, in <module>
    raise Exception('No supported validity device found')
Exception: No supported validity device found
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 153, in apport_excepthook
    with os.fdopen(os.open(pr_filename,
FileNotFoundError: [Errno 2] No such file or directory: '/var/crash/_usr_bin_validity-sensors-firmware.0.crash'

Original exception was:
Traceback (most recent call last):
  File "/usr/bin/validity-sensors-firmware", line 92, in <module>
    raise Exception('No supported validity device found')
Exception: No supported validity device found
```

```
sudo python3 /usr/share/python-validity/playground/factory-reset.py
Traceback (most recent call last):
  File "/usr/share/python-validity/playground/factory-reset.py", line 5, in <module>
    usb.open()
  File "/usr/lib/python3/dist-packages/validitysensor/usb.py", line 50, in open
    self.open_dev(dev)
  File "/usr/lib/python3/dist-packages/validitysensor/usb.py", line 62, in open_dev
    raise Exception('No matching devices found')
Exception: No matching devices found
```

Restarted the service but still not working. 
```
sudo systemctl start python3-validity
fprintd-enroll
list_devices failed: 
```

uunicorn, 2022-07-26:
Hi @amikot , 

`python-validity` is not a generic/universal driver. It is only intended to support some subset of Synaptic (used to be called Validity) sensors which share the same low-level protocol.

You also mentioned `fprintd` device list. But `python-validity` is not supposed to work with `fprintd`.  In fact `python-validity` + `open-fprintd` packages are mutually exclusive with `libfprint` + `fprintd`. If your sensor is listed as supported by `fprintd`, I suggest you completely remove `python-validity` and `open-fprintd` and install `libfprint` and `fprintd` instead.

I hope it makes sense.

amikot, 2022-07-26:
Hi @unicorn,

Okay, now I understand - has been confused by video-tutorial 
[https://www.youtube.com/watch?v=4rWYXc4F-jU](https://www.youtube.com/watch?v=4rWYXc4F-jU)

Which tells people to check compatibility of the sensor with fprintd on their page and if it's listed there, then install python-validity with open-fprintd.
I think the  causes of that incorrect assumption are 1. similar name fprintd and open-fprintd and 2. lack of compatibility list dedicated for python-validity.

It's too bad that so nice solution as python-validity is not taught as universal/generic solution, but have to accept that - what else I can do ?

Anyway - thanks for answer and have nice day :)

uunicorn, 2022-07-26:
> Which tells people to check compatibility of the sensor with fprintd on their page and if it's listed there, then install python-validity with open-fprintd.

It is actually the opposite. In his video he said that you should try to install `python-validity` if it's *not* supported by the `fprintd` project. In his case he had 06cb:009a Synaptics (Validity) sensor, which was not on the supported list of `fprintd`. This is why he installed `python-validity`.

> similar name fprintd and open-fprintd

Yeah, I can see how this could be confusing. `open-fprintd` mimics `fprintd` in a sense that it implements the same fprint DBus interface, so that clients (like Gnome, or PAM modules) can talk to either of them.

> lack of compatibility list dedicated for python-validity

Yep, I agree. Would be nice to have a section in README with a list of supported devices. Here is what `python-validity` sort of supports:
https://github.com/uunicorn/python-validity/blob/master/validitysensor/usb.py#L16