uunicorn/python-validity #147: fprintd-enroll won't actually register fingerprints

bckelley, 2022-11-24:
[image]

So what gives same finger everytime but it fails? I don't understand

me-2k, 2023-01-01:
For me it kept trying to enroll right-index-finger, so I manually entered
`fprintd-enroll --finger left-index-finger`

You can check the result with
`fprintd-list <username>`

And verify by
`fprintd-verify`

sidneijp, 2024-10-28:
@bckelley each time you up and down the indicated finger, try to slightly change its position. Eventually it will pass.

As @me-2k pointed out you can pass an option to set which finger you're gonna setup. The choices are:

```
left-thumb
left-index-finger
left-middle-finger
left-ring-finger
left-little-finger

right-thumb
right-index-finger
right-middle-finger
right-ring-finger
right-little-finger
```

For Ubuntu's Users (in my case 24.0.1 w/ GNOME interface) there is a GUI on "Settings" app to do so:

1.
[image]

2. (mine shows "Enabled" because I'm already using it (already registered a finger)
[image]

3.
[image]

P.S.: I've used PPA repository from kinetic (22.10) Ubuntu