uunicorn/python-validity #148: Can't login anymore

bckelley, 2022-11-25:
Shutdown my laptop last night turned it on this morning and the fingerprint reader doesn't working keeps saying 'login failed' please advice

Sesam31, 2022-12-27:
I'm also having the same issue.
I tried a clean reinstall of python validity and successfully re-enrolled the fingerprint but it still always says login failed.

snoogie2k, 2023-01-25:
For me the first login always fails, I have to use my password, after that everything is fine...

snoogie2k, 2023-01-26:
okay, so I found the reason why I got stuck in a "login-loop" when starting the first time:

I have an encrypted home-dir and the fingerprint login does not work for the first time because my password/passphrase is needed to decyrpt my home... When I logged in for the first time successfully I can use the fingerprint reader just fine.

CosmicChrisss, 2023-04-09:
> okay, so I found the reason why I got stuck in a "login-loop" when starting the first time:
>
> I have an encrypted home-dir and the fingerprint login does not work for the first time because my [...]

I'm in the same boat, did you ever find a workaround?