uunicorn/python-validity #151: fprintd-enroll goes directly to enroll failed

kp4djt, 2023-01-08:
Just installed the code on my Thinkpad X1 Carbon Gen 4. Running Mate Ubuntu 22.04.
Here is what I get when I invoke fprintd-enrol
kp4djt@kp4djt-carbon:~$ fprintd-enroll
Using device /net/reactivated/Fprint/Device/0
Enrolling right-index-finger finger.
Enroll result: enroll-failed
kp4djt@kp4djt-carbon:~$ 

It goes immediately to the Enroll result not allowing me to swipe finger/ 
Results of sudo systemctl status python3-validity
● python3-validity.service - python-validity driver dbus service
     Loaded: loaded (/lib/systemd/system/python3-validity.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2023-01-08 01:05:40 CST; 14min ago
   Main PID: 832 (python3)
      Tasks: 3 (limit: 9082)
     Memory: 25.1M
        CPU: 342ms
     CGroup: /system.slice/python3-validity.service
             └─832 python3 /usr/lib/python-validity/dbus-service --debug

Jan 08 01:07:03 kp4djt-carbon python3[832]: DEBUG:root:<tls< 17: 0104
Jan 08 01:07:03 kp4djt-carbon ERROR:root:Failed[832]: 0401
                                                      Traceback (most recent call last):
                                                        File "/usr/lib/python-validity/dbus-service", line 157, in run
                                                          sensor.enroll(usr, index, update_cb)
                                                        File "/usr/lib/python3/dist-packages/validitysensor/sensor.py", line 828, in enroll
                                                          self.create_enrollment()
                                                        File "/usr/lib/python3/dist-packages/validitysensor/sensor.py", line 749, in create_enrollment
                                                          assert_status(tls.app(pack('<BL', 0x69, 1)))
                                                        File "/usr/lib/python3/dist-packages/validitysensor/util.py", line 12, in assert_status
                                                          raise Exception('Failed: %04x' % s)
                                                      Exception: Failed: 0401
Jan 08 01:07:03 kp4djt-carbon python3[832]: DEBUG:root:EnrollStatus

I have other code running Python3 so not sure what to think.

Hansxsourse, 2023-01-11:
Same issue with same platform

zgshen, 2023-01-13:
Same issue on T460p running Ubuntu 22.04.

tym0niasty, 2023-06-18:
The same error T460p with Ubuntu 22.04.02 LTS.

In the GUI I can see now the fingerprint section (it was not available before using this package) but the fprintd-enroll returns error.

The same if I try to add fingerprint in GUI:
[image]
It does not even allow me to scan - just once I select scan and pick which finger it displays the above.

StewartPolsky, 2023-08-17:
Same issue T460s Mint 21.2

uunicorn, 2023-08-17:
> Exception: Failed: 0401

This looks suspiciously similar to #71 . Can you check which device you have with `lsusb`?

StewartPolsky, 2023-08-17:
@uunicorn thank you for the quick reply. Verified my device with `lsusb` I have 138a:0090. Looks like it is the same issue as #71 in that I can't enroll as it is match-on-host.

Just trying to figure out how to enroll my fingerprint now

uunicorn, 2023-08-17:
Yeah, as explained in #71 , match-on-host devices are not supported by python-validity. 

There is [a branch](https://github.com/3v1n0/libfprint/blob/vfs0090/README.md) of the official libfprint project which adds partial 138a:0090 support. "Partial" here means that this branch rely on your device to be paired by an external tool. This external tool itself is based on python-validity code to do the pairing. This is why python-validity sort of supports 138a:0090 but not really.

To get it working you can try removing python-validity, open-fprintd and follow instructions from the link above.

StewartPolsky, 2023-08-17:
I had initially started with the branch you linked. Unfortunately the repo seems to be dead. The PPA has not been updated to support Ubuntu 22.04.

When I try to compile myself I receive some errors. There are issues opened for both of these but there doesn’t seem to have been an update in years.

uunicorn, 2023-08-18:
Yeah, neither of these projects receive much love unfortunately. Ages ago I created a branch of python-validity to add a proper support for match-on-host devices (138a:0090 specifically). I wanted it to be compatible with Windows, so that fingers enrolled on Windows will work on Linux and vice versa. I got to a point where I reverse engineered the finger template format stored on the device. I also found a library which Synaptic / Validity software is using (based on?) to enroll / match finger templates. It appears to be a redistributable part of DigitalPersona SDK. I.e. if you buy DigitalPersona SDK, you can redistribute those libraries as part of your product. It appears to be present in a bunch of Android phones and tablets. There are also x86 Linux binaries available. But I don't think we can simply steal this library and put it into an open source project. Re-implementing the whole math behind template matching / generation is way too much work. Besides, it will probably still violate a bunch of IP rights and patents.
So, I just gave up.

e64462, 2026-02-09:
@uunicorn 138a:0090 works on archlinux as of today. Not sure if you have any interest or can port it to Ubuntu. I had it working 2 days ago before uninstalling Arch for unrelated reasons. Doesn't seem anyone has it working on Ubuntu though:

https://gitlab.archlinux.org/gugah/libfprint

Here is the AUR page that I used to build.

https://aur.archlinux.org/packages/libfprint-vfs009x-git

Gugah's comment from last July may also be relevant

```
I was able to merge the current libfprint release from https://gitlab.freedesktop.org/libfprint/libfprint/-/tags/v1.94.9

You may have to delete and enroll the fingerprint again after updating:

fprintd-delete $USER
fprintd-enroll

Also verify it's working with:

fprintd-verify

If you keep having any issues, reset the fingerprint device in your BIOS (under Security -> Fingerprint) and initialize it again using a Windows VM with the [official driver](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-p-series-laptops/thinkpad-p50/downloads/ds120491-synaptics-metallica-moh-touch-fingerprint-reader-driver-for-windows-10-version-1709-or-later-thinkpad?category=Fingerprint%20Reader) (using USB device pass-through) or [validity-sensors-tools-git](https://aur.archlinux.org/packages/validity-sensors-tools-git).

Upstream info:

There haven't been recent updates to [3v1n0's patched libfprint](https://github.com/3v1n0/libfprint) nor the [vfs0090 driver submodule](https://github.com/3v1n0/libfprint-tod-vfs0090) which is also hosted in https://gitlab.freedesktop.org/3v1n0/libfprint-tod-vfs0090.

I therefore forked the [official libfprint](https://gitlab.freedesktop.org/libfprint/libfprint) (https://gitlab.archlinux.org/gugah/libfprint) and patched it with a forked driver https://gitlab.archlinux.org/gugah/libfprint-tod-vfs0090

The new patched driver includes some [fixes](https://gitlab.archlinux.org/gugah/libfprint-tod-vfs0090/-/compare/6084a1545589beec0c741200b18b0902cca225ba...master?from_project_id=99376) from [3v1n0's original one](https://github.com/3v1n0/libfprint-tod-vfs0090):

    https://github.com/3v1n0/libfprint-tod-vfs0090/compare/master...Shade30:libfprint-tod-vfs0090:master (credits to svfedorov)
    https://gitlab.freedesktop.org/3v1n0/libfprint-tod-vfs0090/-/merge_requests/2/diffs (credits to valodim)
    https://gitlab.archlinux.org/gugah/libfprint-tod-vfs0090/-/merge_requests/1/diffs
```