uunicorn/python-validity #165: Can't install on Manjaro

Dominux, 2023-07-18:
<details>
<summary>Installing logs</summary>

```bash
==> Making package: open-fprintd 0.6-3 (Tue 18 Jul 2023 07:15:32 PM MSK)
==> Retrieving sources...
==> Making package: python-validity 0.14-1 (Tue 18 Jul 2023 07:15:32 PM MSK)
==> Retrieving sources...
  -> Downloading open-fprintd-0.6.tar.gz...
  -> Downloading python-validity-0.14.tar.gz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0==> Making package: fprintd-clients 1.90.1.r2.g54e56d6-4 (Tue 18 Jul 2023 07:15:32 PM MSK)
==> Retrieving sources...
  -> Cloning fprintd-clients git repo...
Cloning into bare repository '/home/dominux/.cache/yay/fprintd-clients/fprintd-clients'...
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100 12888    0 12888    0     0  15995      0 --:--:-- --:--:-- --:--:--     0
==> Validating source files with md5sums...
    open-fprintd-0.6.tar.gz ... Passed
100  134k    0  134k    0     0   119k      0 --:--:--  0:00:01 --:--:--  646k
==> Validating source files with md5sums...
    python-validity-0.14.tar.gz ... Passed
remote: Enumerating objects: 2893, done.
remote: Counting objects: 100% (1116/1116), done.
remote: Compressing objects: 100% (82/82), done.
remote: Total 2893 (delta 1056), reused 1051 (delta 1029), pack-reused 1777
Receiving objects: 100% (2893/2893), 1.07 MiB | 1.28 MiB/s, done.
Resolving deltas: 100% (2134/2134), done.
  -> Found 0001-disable-systemd-reactivated.diff
  -> Found 0002-Remove-ignored-positional-arguments.patch
==> Validating source files with sha256sums...
    fprintd-clients ... Skipped
    0001-disable-systemd-reactivated.diff ... Passed
    0002-Remove-ignored-positional-arguments.patch ... Passed
:: (1/3) Parsing SRCINFO: python-validity
:: (2/3) Parsing SRCINFO: open-fprintd
:: (3/3) Parsing SRCINFO: fprintd-clients
==> Making package: fprintd-clients 1.90.1.r2.g54e56d6-4 (Tue 18 Jul 2023 07:15:35 PM MSK)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating fprintd-clients git repo...
  -> Found 0001-disable-systemd-reactivated.diff
  -> Found 0002-Remove-ignored-positional-arguments.patch
==> Validating source files with sha256sums...
    fprintd-clients ... Skipped
    0001-disable-systemd-reactivated.diff ... Passed
    0002-Remove-ignored-positional-arguments.patch ... Passed
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Creating working copy of fprintd-clients git repo...
Cloning into 'fprintd-clients'...
done.
Switched to a new branch 'makepkg'
==> Starting prepare()...
Note: switching to '54e56d660bf9730f8abc8f2c6a358ef2fec675f2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 54e56d6 Make building the daemon optional
/home/dominux/.cache/yay/fprintd-clients/PKGBUILD: line 26: patch: command not found
==> ERROR: A failure occurred in prepare().
    Aborting...
 -> Failed to install layer, rolling up to next layer.error:error making: fprintd-clients - exit status 4
==> Making package: open-fprintd 0.6-3 (Tue 18 Jul 2023 07:15:37 PM MSK)
==> Checking runtime dependencies...
==> Missing dependencies:
  -> fprintd-clients
==> Checking buildtime dependencies...
==> ERROR: Could not resolve all dependencies.
 -> Failed to install layer, rolling up to next layer.error:error making: open-fprintd - exit status 8
==> Making package: fprintd-clients 1.90.1.r2.g54e56d6-4 (Tue 18 Jul 2023 07:15:38 PM MSK)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating fprintd-clients git repo...
  -> Found 0001-disable-systemd-reactivated.diff
  -> Found 0002-Remove-ignored-positional-arguments.patch
==> Validating source files with sha256sums...
    fprintd-clients ... Skipped
    0001-disable-systemd-reactivated.diff ... Passed
    0002-Remove-ignored-positional-arguments.patch ... Passed
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Creating working copy of fprintd-clients git repo...
Cloning into 'fprintd-clients'...
done.
Switched to a new branch 'makepkg'
==> Starting prepare()...
Note: switching to '54e56d660bf9730f8abc8f2c6a358ef2fec675f2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 54e56d6 Make building the daemon optional
/home/dominux/.cache/yay/fprintd-clients/PKGBUILD: line 26: patch: command not found
==> ERROR: A failure occurred in prepare().
    Aborting...
 -> error making: fprintd-clients-exit status 4
==> Making package: open-fprintd 0.6-3 (Tue 18 Jul 2023 07:15:39 PM MSK)
==> Checking runtime dependencies...
==> Missing dependencies:
  -> fprintd-clients
==> Checking buildtime dependencies...
==> ERROR: Could not resolve all dependencies.
 -> error making: open-fprintd-exit status 8
==> Making package: python-validity 0.14-1 (Tue 18 Jul 2023 07:15:40 PM MSK)
==> Checking runtime dependencies...
==> Missing dependencies:
  -> open-fprintd>=0.6
  -> fprintd-clients
==> Checking buildtime dependencies...
==> ERROR: Could not resolve all dependencies.
 -> error making: python-validity-exit status 8
 -> Failed to install the following packages. Manual intervention is required:
open-fprintd - exit status 8
python-validity - exit status 8
fprintd-clients - exit status 4

```
</details>

Which type information is also u need to provide

Coriakin, 2023-07-21:
Did you try installing this via the AUR?

I'm on EndevourOS and have had little/no issues for several weeks from first install, except the service needing to be restarted upon laptop resume from sleep state.

I cannot remember if I manually installed those packages it says it depends on, because they are install on my machine.

Did you try to install those?
- open-fprintd
- fprintd-clients

dblast-x, 2023-08-11:
Hello, I'm trying to install the whole python validity with 'yay' in garuda-linux, but i don't get to delete fprintd from the system:: 
    removing fprintd breaks dependency 'fprintd' required by input-devices-support.
I'm actually new to to arch, and I don't seem to find a good way yet.