uunicorn/python-validity #167: "Failed to match fingerprint" even after successfully enrolling fingerprints

sm-wilson, 2023-08-10:
As the title says, I installed without issues on Ubuntu 22.04.2 on a Thinkpad T480s. Tried enrolling fingerprints via both the terminal and through Gnome settings and it seems to be successful, but trying to authenticate does not work and says failed to match fingerprint. FWIW, this is a fresh Ubuntu install, I had fingerprint authentication working fine a year or so ago on a previous point release of 22.04.

Goldfels, 2023-09-16:
What worked for me is factory resetting the device according to these steps: https://github.com/uunicorn/python-validity#errors-on-startup

I've copy-pasted the commands here too:

```
$ sudo systemctl stop python3-validity
$ sudo validity-sensors-firmware
$ sudo python3 /usr/share/python-validity/playground/factory-reset.py

# At some of the above points you may get a 'device busy' error,
# depending on how systemctl plays along. Kill offending processes if
# necessary, or re-run the systemctl stop python3-validity command, 
# in case it has automatically been restarted, or or kill other
# offending processes.

$ sudo systemctl start python3-validity
$ fprintd-enroll
```

sm-wilson, 2024-01-14:
Sorry for the super late reply, I did another tour of distro hopping on this laptop and came back to Ubuntu again and am still having this issue. I'm not able to run the factory reset script, and it's not exactly 'device busy' but 'resource busy.' As such, I don't see any processes mentioned to kill. Here's the output I see after trying to run it:

`Traceback (most recent call last):
  File "/usr/share/python-validity/playground/factory-reset.py", line 6, in <module>
    factory_reset()
  File "/usr/lib/python3/dist-packages/validitysensor/sensor.py", line 85, in factory_reset
    assert_status(usb.cmd(reset_blob))
  File "/usr/lib/python3/dist-packages/validitysensor/usb.py", line 103, in cmd
    self.dev.write(1, out)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 986, in write
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 229, in setup_request
    self.managed_claim_interface(device, intf)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/usb/core.py", line 178, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 829, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/lib/python3/dist-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy`

qwmhq, 2024-02-07:
I had the same "resource busy" error when i tried to run the playground scripts on my Thinkpad T480. I got rid of the error by killing the process using the fingerprint reader.

Here's the steps I took:
Run the command `lsusb`, and look for the line with "Fingerprint Reader". 
I got something like this:
`Bus 001 Device 005: ID XXXX:XXXX Synaptics, Inc. Metallica MIS Touch Fingerprint Reader`

Run the command `sudo lsof /dev/bus/usb/<bus number>/<device number>`
In my case, this was `sudo lsof /dev/bus/usb/001/005`. 

It should give an output similar to this:
`COMMAND   PID USER   FD   TYPE DEVICE SIZE/OFF NODE NAME`
`python3 24395 root    9u   CHR  189,4      0t0  415 /dev/bus/usb/001/005`

Kill the process using `sudo kill -9 <pid>` and try to reset the device again

sm-wilson, 2024-02-09:
Very awesome, that seems to have done the trick! Out of curiosity, is that in the documentation somewhere, or you just knew where to look to find that process to kill?

qwmhq, 2024-02-09:
I figured it out after a few google searches for "how to find process using USB resource". Glad I was able to help you out.