uunicorn/python-validity #172: Cannot install on Fedora 39

latenightdef, 2023-09-26:
Looks like Fedora 39 ships Python 3.12 while python-validity needs Python 3.11, this happens when installing from tigro copr.

```
Last metadata expiration check: 0:33:54 ago on Tue 26 Sep 2023 09:26:14 AM +07.
Package fprintd-clients-1:1.94.2-2.fc39.x86_64 is already installed.
Package fprintd-clients-pam-1:1.94.2-2.fc39.x86_64 is already installed.
Error: 
 Problem 1: conflicting requests
  - nothing provides python(abi) = 3.11 needed by open-fprintd-0.6-7.fc39.noarch from copr:copr.fedorainfracloud.org:tigro:python-validity
 Problem 2: conflicting requests
  - nothing provides python(abi) = 3.11 needed by python3-validity-0.14-2.fc39.noarch from copr:copr.fedorainfracloud.org:tigro:python-validity
  - nothing provides python3.11dist(cryptography) >= 2.1.4 needed by python3-validity-0.14-2.fc39.noarch from copr:copr.fedorainfracloud.org:tigro:python-validity
  - nothing provides python3.11dist(pyusb) >= 1 needed by python3-validity-0.14-2.fc39.noarch from copr:copr.fedorainfracloud.org:tigro:python-validity
  - nothing provides python3.11dist(pyyaml) >= 3.12 needed by python3-validity-0.14-2.fc39.noarch from copr:copr.fedorainfracloud.org:tigro:python-validity
(try to add '--skip-broken' to skip uninstallable packages)
```

Jeannok, 2023-10-05:
Hello;

Exactly the same problem here.

KayJay7, 2023-11-08:
Same problem

Mintu2002, 2023-11-09:
Same problem, waiting for fix

Fulgurance, 2023-11-13:
I have the same problem, how can I fix that ?

Mintu2002, 2023-11-13:
> I have the same problem, how can I fix that ?

I've seen that you have to compile it by yourself

vasi, 2023-11-13:
I forked the COPR here, and it seems to work: https://copr.fedorainfracloud.org/coprs/djvasi/python-validity/  I make no commitment to continue to maintain this though.

appelgriebsch, 2023-11-13:
> I forked the COPR here, and it seems to work: https://copr.fedorainfracloud.org/coprs/djvasi/python-validity/ I make no commitment to continue to maintain this though.

hmm still says `python(abi) = 3.11 needed ...` on my Fedora Silverblue 39?! It also seem to depend on an older version of fprintd-clients: it wants to have fprintd-clients < 1.94.2-2 and on the system I have 1.94.2-8....

wmichaelperry, 2023-11-13:
> I forked the COPR here, and it seems to work: https://copr.fedorainfracloud.org/coprs/djvasi/python-validity/ I make no commitment to continue to maintain this though.

This works on my Thinkpad X1 Yoga with Fedora 39 if "sudo dnf copr enable djvasi/python-validity" is used in place of "sudo dnf copr enable tigro/python-validity" prior to installation.

Mintu2002, 2023-11-13:
> I forked the COPR here, and it seems to work: https://copr.fedorainfracloud.org/coprs/djvasi/python-validity/ I make no commitment to continue to maintain this though.

Works on P52

appelgriebsch, 2023-11-13:
weird.... clearly broken on Silverblue 39
```
Problem: conflicting requests
  - package python3-validity-0.14-2.fc39.noarch from copr:copr.fedorainfracloud.org:djvasi:python-validity requires open-fprintd, but none of the providers can be installed
  - package open-fprintd-0.6-7.fc39.noarch from copr:copr.fedorainfracloud.org:djvasi:python-validity requires fprintd-clients, but none of the providers can be installed
  - package fprintd-clients-1:1.94.2-2.fc39.x86_64 from copr:copr.fedorainfracloud.org:djvasi:python-validity obsoletes fprintd < 1:1.94.2-2.fc39 provided by fprintd-1.94.2-8.fc39.x86_64 from @System
  - nothing provides python(abi) = 3.11 needed by open-fprintd-0.6-7.fc39.noarch from copr:copr.fedorainfracloud.org:djvasi:python-validity
  - nothing provides python(abi) = 3.11 needed by python3-validity-0.14-2.fc39.noarch from copr:copr.fedorainfracloud.org:djvasi:python-validity
  - nothing provides python3.11dist(cryptography) >= 2.1.4 needed by python3-validity-0.14-2.fc39.noarch from copr:copr.fedorainfracloud.org:djvasi:python-validity
  - nothing provides python3.11dist(pyusb) >= 1 needed by python3-validity-0.14-2.fc39.noarch from copr:copr.fedorainfracloud.org:djvasi:python-validity
  - nothing provides python3.11dist(pyyaml) >= 3.12 needed by python3-validity-0.14-2.fc39.noarch from copr:copr.fedorainfracloud.org:djvasi:python-validity
```

bejuba, 2023-11-15:
@appelgriebsch on silverblue you have to manually override some packages first:

`sudo rpm-ostree override remove fprintd fprintd-pam`

guillome-mgn, 2023-11-15:
> I forked the COPR here, and it seems to work: https://copr.fedorainfracloud.org/coprs/djvasi/python-validity/ I make no commitment to continue to maintain this though.

Works on my ThinkPad T470, thank you :+1:

appelgriebsch, 2023-11-15:
> @appelgriebsch on silverblue you have to manually override some packages first:
> 
> `sudo rpm-ostree override remove fprintd fprintd-pam`

ahhh nah.... it still tries to access the readonly /usr/share directory. I thought this has been solved long ago?!

[image]

GingerGraham, 2023-12-11:
> hmm still says `python(abi) = 3.11 needed ...` on my Fedora Silverblue 39?! It also seem to depend on an older version of fprintd-clients: it wants to have fprintd-clients < 1.94.2-2 and on the [...]

I actually see this issue even with the djvasi COPR on F39 (not Silverblue)

kambiz, 2024-02-05:
> I actually see this issue even with the djvasi COPR on F39 (not Silverblue)

I too have the same issue with the djvasi repository mirror. Any updates on this?

jibsaramnim, 2024-02-06:
> I too have the same issue with the djvasi repository mirror. Any updates on this?

In case it helps: I don't know who has created this particular copr, but I was curious to try to see if I could get the fingerprint to work with my just-acquired T480, and with [sneexy/python-validity](https://copr.fedorainfracloud.org/coprs/sneexy/python-validity/) I was able to successfully install and set it up on Fedora Silverblue 39.

Not directly related to copr shenanigans (I believe), but mentioning it just in case: In my case I also had to edit `open-fprintd-resume.service` to have it also restart the `python3-validity` service, as without that it wouldn't work post-suspend.

GingerGraham, 2024-03-13:
> In case it helps: I don't know who has created this particular copr, but I was curious to try to see if I could get the fingerprint to work with my just-acquired T480, and with [...]

Thanks for this! The [sneexy/python-validity](https://copr.fedorainfracloud.org/coprs/sneexy/python-validity/) copr worked perfectly for me too!

PVasileff, 2025-04-30:
> Thanks for this! The [sneexy/python-validity](https://copr.fedorainfracloud.org/coprs/sneexy/python-validity/) copr worked perfectly for me too!

Thanks. This works correctly on Fedora 42 with KDE on a Lenovo T480.