uunicorn/python-validity #174: Impossible to enroll: GDBus.Error:net.reactivated.Fprint.error.NoSuchDevice

JonathanTRisoldi, 2023-10-03:
Neofetch:
---------- 
OS: Fedora Linux 38 (Workstation Edition) x86_64 
Host: 20L5000YUS ThinkPad T480 
Kernel: 6.2.9-300.fc38.x86_64 
Uptime: 24 mins 
Packages: 1878 (rpm) 
Shell: bash 5.2.15 
Resolution: 1920x1080 
DE: GNOME 44.0 
WM: Mutter 
WM Theme: Adwaita 
Theme: Adwaita [GTK2/3] 
Icons: Adwaita [GTK2/3] 
Terminal: gnome-terminal 
CPU: Intel i5-8250U (8) @ 3.400GH
GPU: Intel UHD Graphics 620 
Memory: 2622MiB / 7816MiB

Sensor ID: 06cb:009a

Error when entering command "fprintd-enroll": Impossible to enroll: GDBus.Error:net.reactivated.Fprint.error.NoSuchDevice

kavemang, 2023-12-03:
same issue, same hardware with Fedora 39

real-jame, 2024-07-27:
Same issue with T480s on Fedora 40

poopsicles, 2024-08-07:
@real-jame have you tried starting python3-validity with `systemctl start python3-validity`? (i also used `enable python3-validity` to make it autostart on startup)

chirmstream, 2024-12-15:
> @real-jame have you tried starting python3-validity with `systemctl start python3-validity`? (i also used `enable python3-validity` to make it autostart on startup)

I had the same issue and this fixed it for me on my T480 Fedora 41