uunicorn/python-validity #179: Support for 06cb:00a8

karthikiyengar, 2023-10-18:
Devices using this: Clevo N150U, Clevo N151ZU, System76 Darter Pro, Tuxedo InfinityBook v5
Prior art: https://github.com/nmikhailov/Validity90/issues/69

Happy to dig around, support, collaborate, code.

byPARSE, 2026-09-08:
### Identification for 06cb:00a8 — it reports sensor type `0x2449`

This device can be identified without pairing or writing anything to it, so
here are concrete numbers for `06cb:00a8`.

**Key point:** `tls.cmd()` transmits in plaintext while no TLS session is open
(that is why `read_tls_flash()` works before `tls.open()`). So `identify_sensor()`
— command `0x75` — can simply be asked directly, no pairing required:

```
--> 0x75 identify_sensor()
    10 bytes, status 0x0000
    0000  00 00 00 00 00 00 6e 00 90 01

    zeroes = 0x00000000   (passes the `if zeroes != 0` sanity check)
    minor  = 0x006e
    major  = 0x0190
```

Feeding that through `dev_info_lookup()` from `validitysensor/hw_tables.py`:

```
DeviceInfo(0x0190, 0x2449, 0x6e, 0xff, '86C  TM3261-003-004')
```

So **06cb:00a8 is sensor type `0x2449`**. `sensor.py` only handles `0x199` and
`0x00db`, which is why adding the USB ID alone cannot work — it would fail at
`Device %s is not supported (sensor type 0x%x)`.

Worth noting for anyone tempted to reason from neighbouring firmware versions:
the `minor` values are firmware build numbers, not a hardware ordering. Around
ours: `0x6a`→`0x199`, `0x6b`→`0x191`, `0x6d`→`0x0000`, `0x6e`→`0x2449`,
`0x6f`→`0x581`. Adjacency means nothing.

### The device speaks the protocol fine

All plaintext informational commands are answered with status `0x0000`:

```
--> 0x01   RomInfo.get()            38 bytes, status 0x0000
    0000  00 00 10 b6 04 54 9a 00 00 00 06 14 01 33 00 01
    0010  00 00 4e 4a 49 00 48 1d 00 21 00 00 00 00 01 00
    0020  00 00 00 00 00 06

--> 0x19   (part of send_init)      68 bytes, status 0x0000
    0000  00 00 02 00 21 00 10 01 00 00 00 00 00 00 00 00
    0010  00 00 00 00 XX XX XX XX 00 00 00 00 XX XX XX XX
    0020  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    (XX = redacted, these two words look device-unique)

--> 0x3e   get_flash_info()         16 bytes, status 0x0000
    0000  00 00 ff 00 ff 00 00 10 10 00 00 00 01 00 00 00

--> 0x4302 get_fw_info()             2 bytes, status 0x04af
```

The `0x04af` on `get_fw_info()` is just the `fwext is not loaded` / "clean
slate" state that `send_init()` already handles — not an incompatibility.

### What is actually missing for `0x2449`

I checked the tables rather than guessing:

| Piece | Status |
|---|---|
| `SensorTypeInfo` (geometry: `bytes_per_line`, `line_width`, `lines_per_calibration_data`) | 27 types present, **`0x2449` absent** |
| `SensorCaptureProg` (capture program) | 28 `dev_type` values present, **`0x2449` absent** |
| `calibration_blob` | likely a non-issue — 7 of the 8 types with geometry `120/112` share one identical blob; only `0x00e4` differs |

So the blocker is the **capture program for type `0x2449`**, plus the geometry
[... 118 lines cut ...]
if ident and len(ident) >= 10:
    body = ident[2:]
    zeroes = int.from_bytes(body[0:4], "little")
    minor = int.from_bytes(body[4:6], "little")
    major = int.from_bytes(body[6:8], "little")
    print("identify_sensor: zeroes=0x%08x  minor=0x%04x  major=0x%04x" % (zeroes, minor, major))
    try:
        from validitysensor.hw_tables import dev_info_lookup
        print("dev_info_lookup ->", dev_info_lookup(major, minor))
    except Exception:
        print("(python-validity not importable here - look up (major, minor) "
              "in validitysensor/hw_tables.py manually)")
else:
    print("0x75 was not answered in plaintext on this device.")

lib.libusb_release_interface(handle, IFACE)
lib.libusb_close(handle)
print("\nDone. Nothing was written to the device.")
```

</details>

Run with `sudo python3 validity_probe.py`. If several `0x2449` devices turn out
to share one firmware build, that at least narrows down what a future
`0x2449` table entry would have to cover.

byPARSE, 2026-09-08:
### Follow-up on 06cb:00a8 (type `0x2449`): the driver tables are extractable, plus a wall

Since my last comment I went through the Windows driver. Summary: the tables
are in there in cleartext, I extracted what `0x2449` needs, and then hit a
different, more fundamental problem. Everything below was verified on hardware
(Clevo N151ZU); nothing was ever written to the sensor's flash.

#### The driver is publicly downloadable

`synaWudfBioUsb.dll` v5.5.2707.1073 (PE32+ x86-64), from the Wortmann TERRA
Mobile 1550 fingerprint package. Its INF (`synaWudfBioUsbClevoProd.inf`) lists
`PID_00A8` explicitly:

```
https://webftp.wortmann.de/dokumentenmanagement_wag/NB_TERRA-MOBILE_1550_FINGERPRINT/02_FingerPrint_8319A977-8354-47C5-A88B-77F5AB75DA4B.zip
```

#### Gotcha: blobs are stored without the leading command byte

I first concluded the init blobs were not in the DLL. That was wrong — I was
searching including the `0x06` prefix that `blobs_*.py` carries. Search for
`blob[1:]` instead and `blobs_90.init_hardcoded[1:]` (484 bytes) matches
verbatim at file offset `0x1b8ed0`.

#### Module table

At file offset `0x15e004`, **66 records of 80 bytes**, uint32 fields:

| offset | meaning |
|---|---|
| f02 | module kind, a **bitmask** (e.g. `0x100050` = `0x100000\|0x50`) |
| f04 | generation, 2 or 3 |
| f08 | payload length |
| f09/f10 | uint64 VA of the payload |
| f15 | ROM major (always 6 here) |
| f16 | ROM minor |
| f05/f07/f17 | `0xffff` wildcards |

Anchoring on the known 0090 blobs maps two kinds:
**`f02=0x004000` = `init_hardcoded`**, **`f02=0x000004` = `reset_blob`**.

#### Capture program and geometry for 0x2449

The `SensorCaptureProg` table lives separately (records of 64 bytes at
`0x1eca04`, blob lists as packed 10-byte entries `uint16 length; uint64 VA`,
length first — hence unaligned pointers). Records #31/#32 are `dev_type=0x2449`
for ROM `6/0x14` and `6/0x15`, blobs `[224, 872, 592]`.

The extracted program is **byte-identical to the one for `dev_type=0x0b4d`**
(all three blobs, both ROM versions), so `0x0b4d`'s `SensorTypeInfo`
(`bytes_per_line=160, line_width=144, lines_per_calibration_data=56,
repeat_multiplier=2`) should apply. Chunk `0x2f` gives `lines_2d = 56`.

#### Two USB findings worth knowing

1. `open_dev()` calls `dev.set_configuration()`. On 06cb:00a8 the device stops
   answering afterwards (timeouts on EP `0x81`). Claiming interface 0 without
   reconfiguring works reliably.
2. When the sensor does wedge, `ioctl(USBDEVFS_RESET)` fully recovers it — no
   reboot needed. Verified twice.

#### Where it stops

This device reports RomInfo `major=6, minor=0x14`, and **for ROM minor `0x14`
there is no module with `f02=0x4000`** — no `init_hardcoded` of the kind
`send_init()` expects. That is exactly why borrowing the 0x009a blob fails with
`0x04be`.

I tried all ten modules present for ROM `6/0x14`:

```
all six with f04=3        -> 0x04be   (so this device is the f04=2 generation)
f02=0x000004  ( 6100 B)   -> 0x04be
f02=0x400080  (10500 B)   -> 0x0000   accepted
f02=0x000080  ( 4676 B)   -> 0x04aa   (different error, notably)
f02=0x100050  (11940 B)   -> 0x0000   accepted, then the device reboots
```

`f02=0x400080` loads cleanly and visibly changes state (one byte in the RomInfo
reply flips `0x00` -> `0x0d`). But:

```
0x3e before module upload: JEDEC 0xff:0xff, blocksize=0, partitions=0
0x3e after  module upload: JEDEC 0xff:0xff, blocksize=0, partitions=0
```

**Hypothesis, not established:** this sensor may have no external SPI flash at
all. `flash_ic_table_lookup(0xff, 0xff, 0)` returns `None`, so `get_flash_info()`
raises `Unknown flash IC` and `init_flash()` aborts before its formatting path —
which is why nothing was ever written. If that hypothesis holds, the flash-based
model (partition table, cert store in partition 1, TLS pairing persisted there)
does not map onto this device, and that is a deeper gap than missing tables.

I would be glad to be wrong about that — if anyone knows how the "86C" family
persists certificates and templates, that is the piece that is missing.

byPARSE, 2026-09-08:
### Correction: `0x04af` is not "fwext not loaded"

In my first comment I wrote that the `0x04af` from `get_fw_info()` is "just the
`fwext is not loaded` / clean slate state that `send_init()` already handles —
not an incompatibility". **That was wrong**, and since it could send someone
down the wrong path, here is the correction.

A read-only sweep over every partition, using only `0x43` (get_fw_info) and
`0x40` (read_flash), returns the same status everywhere:

```
0x43 <p>  for p = 0..7   ->  0x04af  (all)
0x40 <p>  read 256 B @0  ->  0x04af  (all)
```

Two things follow. First, `flash.py` already has a distinct return for "no
firmware present" — it checks for `0xb004` and returns `None` — so `0x04af` is
not that. Second, a uniform failure across *all* partitions and *both* a
metadata command and a data read does not look like "this partition has no
firmware yet"; it looks like the flash subsystem is not reachable at all.

Combined with `0x3e` reporting JEDEC `0xff:0xff` and `blocksize=0` (unchanged
before and after a successful module upload), the earlier hypothesis stands
sharper: **this device most likely has no external SPI flash**, so it cannot be
the "clean slate that just needs initialising" I implied.

#### Why there is nothing to copy from

Worth recording, because it was my next idea and it does not work: both types
python-validity supports run an old firmware generation, while this device does
not.

```
supported  dev_type 0x0199  ->  RomInfo 6/0x00
supported  dev_type 0x00db  ->  RomInfo 6/0x00
this one   dev_type 0x2449  ->  RomInfo 6/0x14
```

And `0x2449` is the only type in `hw_tables.py` carrying the `86C` name prefix
(`0x0199` is `57K0`, `0x00db` is `55E`/`VSI`), so there is no sibling type whose
behaviour could be reused either.

So the gap is not a missing table — the storage architecture changed between
these generations. Finding out how the `86C` family persists certificates and
templates means disassembling the driver's *code*, not just lifting its data
tables, and there the "anchor on known bytes" trick that made the table
extraction easy no longer applies.

byPARSE, 2026-09-08:
### Architecture: why neither python-validity nor libfprint's `synaptics` fits 06cb:00a8

Two negative results and one vendor-sourced explanation, so nobody has to
re-walk these.

#### The driver INF names the device, and a sibling

Reading `synaWudfBioUsbClevoProd.inf` (UTF-16, strip the NULs) gives:

```
%WBDIDeviceName% = Biometric_Install, USB\VID_06CB&PID_0078
%WBDIDeviceName% = Biometric_Install, USB\VID_06CB&PID_00A8

WBDIDeviceName  = "Synaptics WBDI Fingerprint Reader - USB 073"
CustomerProject = "Clevo-073"
```

So **06cb:0078 is a sibling** covered by the same driver, and the vendor's own
name for this part is "USB 073". For the record, `06cb:00a8` is *not* the
"Metallica MOH Touch" — that is `06cb:00a2`.

This device family has been asked about before: a
[January 2018 fprint mailing list thread](https://lists.freedesktop.org/archives/fprint/2018-January/000918.html)
about `06cb:0078` on a Clevo P751DM2-G, answered with "From what I can tell,
it's not supported."

#### Why there is no external flash

Synaptics' own material on the Match-in-Sensor line states that "fingerprint
image enrollment, pattern storage and biometric matching" happen *inside* the
sensor, that the sensor has "its own internal flash memory", and that — unlike
match-on-host — the host never receives fingerprint data.

That fits what the device reports exactly: `0x3e` returning JEDEC `0xff:0xff`
with `blocksize=0` is what you get when there is no external SPI part to answer
the ID read at all. So the flash-partition model python-validity is built on
(cert store in partition 1, template DB in partition 4, TLS state persisted
there) has nothing to attach to on this device.

#### But it is not a BMKT device either

The obvious next thought is libfprint's `synaptics` driver, which speaks BMKT
and does in-sensor enroll/verify with no image capture and no flash handling.
That was my hypothesis; it is wrong. Framing a BMKT message as
`FE <seq> <msg_id> <payload_len>` and sending it:

```
FE 01 B1 00   (GET_VERSION)   ->  01 04
FE 00 B1 00   (GET_VERSION)   ->  01 04
FE 01 A1 00   (FPS_INIT try)  ->  01 04
0x01          (Validity)      ->  38 bytes, status 0x0000
```

`0104` is status `0x0401`, i.e. a *Validity* protocol error reply to an unknown
command — not a BMKT response. The Validity command still works immediately
afterwards. So this sensor speaks the Validity protocol, not BMKT, and adding
`0x00a8` to the `synaptics` driver's `id_table` would not help.

#### Summary

`06cb:00a8` sits between the two supported worlds: Validity command protocol
like the older parts, but no external flash like the newer ones. Its RomInfo is
`6/0x14` where both python-validity-supported types are `6/0x00`.

If that reading is right, supporting it is not a matter of tables or a driver
id — it needs whatever command set this generation uses to enroll and match
against its internal storage. I have not found that, and I would welcome being
told it exists somewhere.

byPARSE, 2026-09-09:
### 06cb:00a8 (ROM 6/20, sensor type `0x2449`): TLS session established, sensor streams live image data

Substantial progress, plus **a correction to something I posted earlier in this thread**. Everything below was done against the real device; nothing was ever written to flash and `0x3f` was never sent.

---

#### Correction first: my "Match-in-Sensor" claim was wrong

Earlier I argued from Synaptics marketing material that this sensor does matching internally and therefore never hands images to the host, and I used that to explain the missing external flash. **That explanation does not hold.** The observation stands — there is no external SPI flash — but the conclusion was wrong:

* `0x2449` is in the sensor-type whitelist in `scsGetPrintCmdHelpers.c`, and that branch performs per-pixel arithmetic on a host-side raw buffer.
* Forward call-graph analysis from `vfmCapture.c`/`vfmAuth.c` reaches `vcsImage.c`, `vcsTemplate.c`, `vcsMatcher.c` and `vcsmQuantumMatcher.c` — a device that matched internally would need none of those.
* And empirically: the host does receive the data (below).

Sorry for the noise. This device is **match-on-host**.

---

#### The Global Wrapping Key is the same for this generation

`scsSharedSecret.c` builds a 32-byte secret from four 8-byte constants emitted by code (`FUN_180073b80/_c70/_d60/_e50`, each with a decoy checksum loop), then runs `PRF(secret, "GWK" + hw_key, 0x20)`.

The value is **byte-identical to `password_hardcoded` in `validitysensor/tls.py`**, and the derivation matches line 116 exactly. The labels `GWK`, `GWK_SIGN`, `master secret` and `key expansion` are all present in the DLL.

Note for anyone searching: these constants are **not stored as contiguous bytes** — they are assembled at runtime, so byte-pattern searches miss them. Look for the generator functions.

---

#### Pairing works without flash

`read_tls_flash()` is a dead end here (`0x40` → `0x04af`), but `init_flash()` already contains the alternative: **command `0x50` returns the ECDH parameters** (`tls.handle_ecdh(rsp[-400:])`). It is gated behind command `0x4f`.

`0x4f` turns out to be safe on this device. Its handler (`FUN_180080d60`, `scsFwV4.c`) builds sections conditionally:

```
base 5 B                        "4f 00 00 00 00"
if [dev+0x27c]==1:
    if [dev+0x274]==0:  section 0 (flash params, 0x0c B)
                        section 1 (partition table, 0xf0 B)
    section 5: client certificate
    if [param+0x50]==1 && [dev+0x274]==0:  section 3 (0x1a4 B)
```

Both flags come from **one capability byte** in the RomInfo payload at offset `0x17`:
`[dev+0x274] = (byte & 0x02)==0`, `[dev+0x27c] = (byte & 0x40)==0`.

On 06cb:00a8 that byte is **`0x21`**, so all three flash-related sections are omitted. The packet is just header + certificate, 453 B total. `make_cert()` from `init_flash.py` is usable unchanged — the 444 B it produces are exactly `184 (cert) + 4 (sig length) + 256 (signature)`, which also resolves the `FIXME not sure this math is right` comment there.

**Working sequence** (each step status 0):

```
06 + reset_blob (5796 B, ROM 6/20 gen 2 module)
4f + with_hdr(5, make_cert(pub))        -> 298 B, device certificate
50                                       -> 926 B, last 400 B = ECDH blob
handle_priv(encrypt_key(priv, pub))
tls.open()                               -> secure_tx = secure_rx = True
```

Caveat: `handle_ecdh()` raises `InvalidSignature` here because the firmware public key it validates against does not match ROM 6/20 (there is a `TODO` about this in the source). I skipped the check to get a session; a production driver should not.

Everything is **volatile** — no flash means the whole sequence must run on every session start. A `USBDEVFS_RESET` returns the device to plaintext state afterwards.

---

#### Two over-restrictions in python-validity

1. `build_cmd_02` raises unless `rom_info.product == 0x30`. We report `0x33`. The vendor driver's lookup (`FUN_18005d810`) maps the product codes `+ - 0 2 3` **all to the same value**, i.e. it treats them identically. Allowing `0x2b/0x2d/0x30/0x32/0x33` is correct.
2. `line_update_type_2` looks for a register write to `0x800020fc` that this capture program does not contain; `line_update_type_1` needs only `0x8000203c`. `0x2449` belongs in `line_update_type1_devices`.

---
[... 31 lines cut ...]
off  0  u16 = 0xb2fb        off 20  u32 = 0x8001
off  2  u8  = 0x0f          off 24  u32 = 0x20a
off  3  u8  = 0x00          off 28  u32 = 0x190b
off  4  u16 = 0x0000        off 32  u16 = 10000 if bit set else 50000
off  6  u8  = 0x0f          off 34  u16 =  6000 if bit set else 60000
off  7  u8  = 0x00          off 36  u8  = 1
off  8  u32 = 0x30          off 37  u8  = 9
off 12  u32 = iVar1 + 0x20000   off 38  u8  = 0x10
off 16  u32 = iVar1 + 0x9ffe0   off 39  u8  = 0
                                (sensor type 0xed: off 16 = 0xa000f)

iVar1 = (100 - [dev+0x238]) * ([dev+0x234] - [dev+0x230]) / 100 + [dev+0x230]
```

Cross-check against the blob currently in `sensor.py`: its off-12 value `0x20087` and off-16 value `0xa0067` both independently yield **iVar1 = 135**, and the constants at off 0/8/20/24/28 match exactly. Only off 32/34 differ (5000/3000 there) — presumably an older DLL.

---

#### Where it still stops

`cmd 02 (IDENTIFY)` is accepted and the **start interrupt (type 0) arrives**, but the **finger-detect interrupt (type 2) never does**, with a finger on the sensor, tested with `iVar1` = 135 and 512. So the threshold alone i [... cut ...] sorDeviceInit.c` / `scsWindsorCal.c`) rather than guessing, and to decode chunk `0x2e` (image reconstruction) — candidate `FUN_18008cc90`.

Also still open: no recognizable ridge pattern has been reconstructed yet, only verified live sensor response. And this sensor appears to deliver **one spatial line repeatedly** — within a frame, line-mean stdev is 0.19; only finger movement produces variation (stdev 9.0, up to 25.4).

Happy to share the raw captures or the scripts if anyone wants to dig in.