uunicorn/python-validity #180: https://download.lenovo.com/pccbbs/mobiles/n1mgf18w.exe

ArnoldSchiller, 2023-10-26:
Now I don't know if the above driver would not work as well. File "/usr/bin/validity-sensors-firmware", line 59, in download_and_extract_fw

    raise Exception('Hash mismatch for driver download!

It would be the driver from 3/23/2023 from Lenovo and in the ppa the last update was in 2022. 

But thanks for the tool - it works also with the older driver.

uunicorn, 2023-10-26:
Hi Arnold, the only thing `python-validity` is using from the Windows driver is a firmware file. Chances are, the new driver contains the same firmware file. And even if it's not the same, it still may still work fine with `python-validity`.  Can you check and if it does work raise a PR to change the checksum?

ArnoldSchiller, 2023-10-27:
> Hi Arnold, the only thing `python-validity` is using from the Windows driver is a firmware file. Chances are, the new driver contains the same firmware file. And even if it's not the same, it may still still work fine with `python-validity`. Can you check and if it does work raise a PR to change the checksum?

Ok the firmware in the update from 23.3.23 is probably a different one - I have now only commented out the hash and then comes

`
  File "/usr/bin/validity-sensors-firmware", line 71, in download_and_extract_fw
    raise Exception('No {} found in the archive'.format(fwname))
Exception: No 6_07f_lenovo_mis_qm.xpfwext found in the archive
`

ArnoldSchiller, 2023-10-27:
Ok close ticket FirmwareVersion = 6.07.0164
ist the same only the Filename changend with a new signature.

ArnoldSchiller, 2023-10-27:
Thx a lot - something learned. Ticket closed.

closed 2023-10-27 by ArnoldSchiller