uunicorn/python-validity #182: PPA doesn't include packages for Ubuntu 23.04 (lunar) or Ubuntu 23.10 (mantic).

Flimm, 2023-11-16:
This is the PPA referenced in the README.md of this repo: https://launchpad.net/~uunicorn/+archive/ubuntu/open-fprintd

It doesn't seem to include packages for Ubuntu 23.04 (lunar lobster) or Ubuntu 23.10 (mantic minotaur).

When running this command:

    sudo add-apt-repository ppa:uunicorn/open-fprintd

I get this error:

    E: The repository 'https://ppa.launchpadcontent.net/uunicorn/open-fprintd/ubuntu mantic Release' does not have a Release file.

sapkyy, 2023-11-19:
Same issue. Any plans to add mantic?

T3d4, 2023-11-23:
Same issue

iamzainmunir, 2023-12-01:
Facing same issue.

barbaldo, 2023-12-20:
Also having this issue, it seems like it's the only way to get the fingerprint sensor to work

BrLex07, 2023-12-27:
Same issue!

Kixel, 2024-01-02:
Facing the same issue. Is there any way to support?

Akross-Nostromo, 2024-01-10:
Same issue!

MrMinimal, 2024-01-20:
I'd also appreciate an update of the PPA

dacog, 2024-03-28:
In this issue https://github.com/uunicorn/python-validity/issues/188 there is a workaround. I just tested it on 23.10 and it works.