uunicorn/python-validity #183: User interrupts screen timeout locking causes python3-validity.service error

ixnewton, 2023-12-17:
If when the screen lock is about to start after screen timeout is interrupted by a mouse/trackpad move python-validity has a problem causing the system to be unresponsive for maybe 30 seconds. The system log shows python3-validity is in some error state.
 
[python3-validity.log](https://github.com/uunicorn/python-validity/files/13695755/python3-validity.log)

ixnewton, 2023-12-19:
Further testing shows that the screen locking setting to allow unlocking without password affects this problem. This can be set to 0 seconds or disabled (KDE plasma desktop) and avoids the problem. 

My suggestion is that if unlocking timer without a password is enabled then python-validity should do nothing until the screen is actually locked?

niko-origin, 2026-10-20:
I am experiencing a similar issue with GNOME/GDM and python-validity.

**System:**
- Lenovo ThinkPad X1 Carbon Gen 5
- CachyOS (Arch-based), GNOME 51.1, GDM 51.1, Wayland
- Validity fingerprint sensor: `138a:0097`
- `python-validity` 0.15-1
- `open-fprintd` 0.7-2
- `fprintd-clients-git` 1.90.1.r2.g54e56d6-10

Fingerprint verification works reliably from a terminal using `fprintd-verify "$USER"`.

However, fingerprint authentication fails intermittently on the GNOME lock screen and at the GDM login screen. GNOME screen unlocking succeeded once, but subsequent attempts failed.

**Failed GNOME unlock (07:01:20):**

```text
Claim
VerifyStart
do_release
Claim
VerifyStart
VerifyStatus
```

At 07:01:28, `python3-validity` logged:

```text
usb.core.USBTimeoutError: [Errno 110] Operation timed out
validitysensor.usb.CancelledException
```

**Successful terminal verification (07:03:45–07:03:47):**

```text
Claim
VerifyStart
VerifyStop
Release
do_release
```

The difference is reproducible: GNOME starts verification twice, releasing the first attempt immediately, whereas terminal verification starts only once and succeeds.

Restarting `python3-validity.service` and `open-fprintd.service` did not fix the issue.

Could the immediate release and restart of verification be related to the USB timeout? Has anyone found a workaround for this behavior with GNOME/GDM?