uunicorn/python-validity #187: Reader works initially on reboot, but stops working after suspend/sleep

matthew28845, 2024-01-19:
I am using a ThinkPad T25, with Validity Sensors 138a:0097 reader. I am able to install successfully, and the fingerprint reader works great initially, but as soon as I put the laptop into sleep mode, it quits working goes completely unresponsive, and will just softlock the computer if I don't have a timeout set for fingeprint auth. 

Here is the output of `systemctl status python3-validity`:
```
● python3-validity.service - python-validity driver dbus service
     Loaded: loaded (/usr/lib/systemd/system/python3-validity.service; enabled; preset: disabled)
     Active: active (running) since Fri 2024-01-19 12:08:39 CST; 3min 18s ago
   Main PID: 919 (python3)
      Tasks: 3 (limit: 18879)
     Memory: 48.8M (peak: 50.5M)
        CPU: 343ms
     CGroup: /system.slice/python3-validity.service
             └─919 python3 /usr/lib/python-validity/dbus-service --debug

Jan 19 12:09:28 MatthewThinkPadT25 python3[919]: DEBUG:root:>cmd> 17030300406d62becfac9e68e95bb89756b09a119d8>
Jan 19 12:09:28 MatthewThinkPadT25 python3[919]: DEBUG:root:<cmd< 1703031040c393ff57b318ff387841de9ce7c769122>
Jan 19 12:09:28 MatthewThinkPadT25 python3[919]: DEBUG:root:>cmd> 170303005039c7a9d856f520b818119eb2315953b62>
Jan 19 12:09:28 MatthewThinkPadT25 python3[919]: DEBUG:root:<cmd< 1703030050098e9e50c7421569a7d88fade2e1d0234>
Jan 19 12:09:28 MatthewThinkPadT25 python3[919]: DEBUG:root:atexit registered
Jan 19 12:09:28 MatthewThinkPadT25 python3[919]: DEBUG:root:In Suspend
Jan 19 12:09:39 MatthewThinkPadT25 python3[919]: DEBUG:root:In Suspend
Jan 19 12:09:48 MatthewThinkPadT25 python3[919]: DEBUG:root:In Resume
Jan 19 12:09:48 MatthewThinkPadT25 python3[919]: DEBUG:root:>cmd> 3e
Jan 19 12:09:48 MatthewThinkPadT25 python3[919]: DEBUG:root:>cmd> 3e
```
This would indicate that the reader should be functioning normally, but as stated I can't get it to actually function. Running `fprintd-enroll` when it's in this state returns this:
```
Using device /net/reactivated/Fprint/Device/0
Enrolling right-index-finger finger.
Enroll result: enroll-failed
```
Services open-fprintd-suspend and open-fprintd-resume are both enabled and started. Any advice would be appreciated. Thanks!

iDrinkCoffee-TG, 2024-02-03:
Issue #106 has a fix with a systemd service.

YovanggaAnandhika, 2024-10-25:
solutions and explanations that can be tried https://github.com/uunicorn/python-validity/issues/128#issuecomment-2437045697