uunicorn/python-validity #188: For Ubuntu 22.10 'Kinetic Kudu', 23.04 'Lunar Lobster' or Ubuntu 23.10 'Mantic Minotaur' this does not work but workaround possible! Please mention this on main page.

Valenteriano, 2024-02-23:
advice for Ubuntu users using Ubuntu 22.10 'Kinetic Kudu', 23.04 'Lunar Lobster' or Ubuntu 23.10 'Mantic Minotaur':

After the commands
- sudo add-apt-repository ppa:uunicorn/open-fprintd
- sudo apt-get update

mentioned in github site, then it tells that it did not find the IP of the repository (or repository itself). My solution for Ubuntu 23.10 'mantic minotaur' was, execute the system app 'software & Updates', go to tab 'Other Software', edit the new entry for fingerprint 'fprintd' (often the last new entry) and write in suites box 'jammy' instead of 'kinetic', 'lunar or 'mantic''.

Then update and continue with the other helpful commands on the github page without enrolling finger. The fingers can be enrolled in user settings (better as correct naming possible). Also important: Do not forget the other helpful settings as under 'Fingerprint not working after waking up from suspend' and so on.

I do not know how lon this workaround works but these distributions work for sure 'tested' (or used) by me long time.

dacog, 2024-03-28:
I can confirm this workaround works on Kubuntu 23.10 on a Thinkpad t480s

I used the following instead of Software and Updates

`nano /etc/apt/sources.list.d/uunicorn-ubuntu-open-fprintd-mantic.sources`

and replace mantic with jammy

bclarke123, 2024-06-02:
Confirming this works on 24.04 Noble as well

vdrover, 2024-06-02:
> Confirming this works on 24.04 Noble as well

Are you saying you have successfully installed python-validity in 24.04? Like others, I am getting the [`does not have release file`](https://github.com/uunicorn/python-validity/issues/191) error in 24.04.

shahzeb-debug, 2024-06-02:
> > Confirming this works on 24.04 Noble as well
> 
> Are you saying you have successfully installed python-validity in 24.04? Like others, I am getting the [`does not have release file`](https://github.com/uunicorn/python-validity/issues/191) error in 24.04.

I am also having the same. The dev hasn't yet submitted the release file. Let's hope that the project is still active.

bclarke123, 2024-06-02:
> > Confirming this works on 24.04 Noble as well
> 
> Are you saying you have successfully installed python-validity in 24.04? Like others, I am getting the [`does not have release file`](https://github.com/uunicorn/python-validity/issues/191) error in 24.04.

Yes, updating the PPA to pull the Jammy source works for me, Lenovo X1C6, device id `06cb:009a` - 

```
$ cat /etc/apt/sources.list.d/uunicorn-ubuntu-open-fprintd-noble.sources
Types: deb
URIs: https://ppa.launchpadcontent.net/uunicorn/open-fprintd/ubuntu/
Suites: jammy
Components: main
Signed-By: -----BEGIN PGP PUBLIC KEY BLOCK-----

[etc]
```

```
$ fprintd-list [me]
found 1 devices
Device at /net/reactivated/Fprint/Device/0
Using device /net/reactivated/Fprint/Device/0
Fingerprints for user [me] on DBus driver (press):
 - #0: right-index-finger
```

```
$ fprintd-verify 
Using device /net/reactivated/Fprint/Device/0
Listing enrolled fingers:
 - #0: right-index-finger
Verify result: verify-match (done)
```

```
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=24.04
DISTRIB_CODENAME=noble
DISTRIB_DESCRIPTION="Ubuntu 24.04 LTS"
```

shahzeb-debug, 2024-06-03:
I can also confirm the solution by @bclarke123 works on Ubuntu 24.04 for 138a:0097 Validity Sensors.

Just replace the suites from "noble" to "jammy" and you are all good to go.

[image]

ToniCipriani, 2024-07-29:
Confirmed working with Linux Mint Wilma, just had to specify `jammy`.

Works great with my Yoga 260 with an 009d swapped over from an X380.

Enetzlsg, 2024-07-29:
> Confirmed working with Linux Mint Wilma, just had to specify `jammy`.
> 

Where do you change this in Mint??

k0nki, 2024-08-02:
You don't need to make any changes to KDE neon User Edition 20240801 to use the python-validity repository. Just follow the instructions for Ubuntu, then register your fingerprints via the user settings.

Works straight out of the box, kudos to the KDE team.

K0nki

sidneijp, 2024-10-28:
I've add the Ubuntu's Kinetic (22.10) PPA repo on my Ubuntu 24.10 (GNOME distro) and it installed and runt just fine.

This OS version is already using the `deb822` standard for apt source files, so to specify the distro version just add `Suites: kinetic` to the respective file:

```shell
# it has 'noble' on its name to 'kinetic' because I didn't renamed it on purpose
vim /etc/apt/sources.list.d/uunicorn-ubuntu-open-fprintd-noble.sources
```
expected contents:

```
Types: deb
URIs: https://ppa.launchpadcontent.net/uunicorn/open-fprintd/ubuntu/
Suites: kinetic  # <----HERE----- it was 'noble' here, so change it for the newest supported distro version (currently 'kinetic')
Components: main
Signed-By: -----BEGIN PGP PUBLIC KEY BLOCK-----
...
```

dacog, 2025-03-09:
I can confirm this workaround works on KDE NEON based on Ubuntu Noble (24.04) on a Thinkpad t480s

just use

`add-apt-repository -y "deb http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu jammy main"`

after installing the packages on KDE Plasma 6 then run:

```shell
sudo pam-auth-update # and select fingerprint
fprintd-enroll # to enroll your fingerprint, or go to settings > users > your user > configure fingerprint authentication
```

dacog, 2025-03-09:
I just used claude.ai to make a summary of this issue and #191 

# Ubuntu Fingerprint Compatibility Table

Below is a summary of all Ubuntu/Linux distro versions where the fingerprint workaround has been confirmed to work, along with the specific fix and users who verified it.

| Distro Version | Codename | Fix | Confirming Users | Count |
|----------------|----------|-----|------------------|-------|
| Ubuntu 22.10 | Kinetic Kudu | Change suite from 'kinetic' to 'jammy' | @Valenteriano | 1 |
| Ubuntu 23.04 | Lunar Lobster | Change suite from 'lunar' to 'jammy' | @Valenteriano | 1 |
| Ubuntu 23.10 | Mantic Minotaur | Change suite from 'mantic' to 'jammy' | @Valenteriano, @dacog, @peachpanic (reported issue) | 3 |
| Kubuntu 23.10 | Mantic Minotaur | Change suite from 'mantic' to 'jammy' | @dacog | 1 |
| Ubuntu 24.04 | Noble Numbat | Change suite from 'noble' to 'jammy' | @bclarke123, @shahxeb-debug, @dacog, @eagle-lk, @grupofaster, @rasa | 6 |
| KDE Neon (Ubuntu 24.04) | Noble | Change suite to 'jammy' or use `add-apt-repository -y "deb http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu jammy main"` | @dacog | 1 |
| Linux Mint | Wilma | Change suite to 'jammy' | @ToniCipriani | 1 |
| KDE Neon User Edition (20240801) | - | No changes needed, works out of the box | @k0nki | 1 |
| Ubuntu 24.10 | - | Change suite from 'noble' to 'kinetic' | @sidneijp | 1 |
| Debian 12 | - | Issue reported, no confirmed fix | @earthseekerx | 0 |
| Fedora 40 | - | Issue reported with traditional method, alternative repo from @taaem suggested | @xkillyridolsx, @shahxeb-debug | 0 |

## Hardware Compatibility Notes
- Confirmed working on ThinkPad T480s (@dacog)
- Confirmed working on ThinkPad T480 (@peachpanic - reported issue, fix not initially confirmed)
- Confirmed working on ThinkPad X1C6, device id 06cb:009a (@bclarke123)
- Confirmed working on ThinkPad Yoga 370 with Validity Sensors 138a:0097 (@shahxeb-debug)
- Confirmed working on ThinkPad Yoga 260 with 009d swapped from X380 (@ToniCipriani)
- Reported but not working initially on ThinkPad X220 (@khatfull)
- Reported issue on ThinkPad T470 with Debian 12 (@earthseekerx)

## Common Method for Fixing 

1. Add the PPA repository:
   ```
   sudo add-apt-repository ppa:uunicorn/open-fprintd
   ```
2. Edit the sources file to change the suite/codename to 'jammy' (or 'kinetic' for 24.10)
3. Update and continue with installation:   ```   sudo apt-get update   ```

### File Location for Editing
- `/etc/apt/sources.list.d/uunicorn-ubuntu-open-fprintd-[distro-codename].sources`

### Command Line Method for Noble (24.04)
Instead of manually editing the file, you can use this command:
```
sudo sed -i 's/noble/jammy/' /etc/apt/sources.list.d/uunicorn-ubuntu-open-fprintd-noble.sources
```

or add the repo for jammy using

 `add-apt-repository -y "deb http://ppa.launchpad.net/uunicorn/open-fprintd/ubuntu jammy main"`

### Alternative for Fedora
For Fedora users, the suggested alternative is to use @taaem's repository:
```
https://copr.fedorainfracloud.org/coprs/taaem/python-validity/
```

### Notes
- Some users reported stability issues after installation (@vasi mentioned login manager crashes occasionally)