uunicorn/python-validity #189: [Ubuntu 22.04] Enroll works, verify does not, no error

Redsandro, 2024-02-29:
Everything works according to instructions. I can install, update PAM, enroll fingers. But the actual reading of fingerprint doesn't work. there is no error message. The fingerprint light just stays on for 10 seconds and doesn't respond to any fingerprints. Then the PAM/login/sudo request reverts to password.

This is `138a:0097 Validity Sensors, Inc.` on a Lenovo ThinkPad X1 Yoga Gen 2 (`20JD000WUS`).

khatfull, 2024-03-27:
I had some goofy problems on a T470 that I just setup.  I could enroll just fine but then never validate, login or sudo.  What appears to have fixed it is using the instructions to factory reset the firmware.  This was on a machine originally setup with Windows 10 and the fingerprint reader, then reloaded with Ubuntu 22.04 (no dual boot).  systemctl status python3-validity did not report errors but since I did have the sensor configured and fingerprints enrolled under Windows I took a flyer at the reset.  Now login and sudo work just fine.

mikpotik, 2024-08-17:
I had the same problem, I used to use Windows 10 and used fingerprint authorization. After I switched to Ubuntu 22.04.4 LTS (Jammy Jellyfish), the fingerprints were registered normally, but the fingerprint login did not work in the login window. I tried to reset according to the author's instructions:

$ sudo systemctl stop python3-validity
$ sudo validity-sensors-firmware
$ sudo python3 /usr/share/python-validity/playground/factory-reset.py

At some of the above points you may get a 'device busy' error, depending on how systemctl plays along. Kill offending processes if necessary, or re-run the systemctl stop python3-validity command,  in case it has automatically been restarted, or or kill other offending processes.

$ sudo systemctl start python3-validity
$ fprintd-enroll

but I ran into a problem with the USB being busy. So I went into Windows, which I had next to Ubuntu on the same system and followed the steps as in the screenshot to remove any previous finger registrations. 

1. Download the file from the link. 
2. Install the file.
3. Run SynapticsUtility.exe
4. Select and Delete everything in the red window by clicking “Delete fingerprints”.
[image]