uunicorn/python-validity #191: Repository does not have release file

yoenuts, 2024-03-29:
Ubuntu Version is 23.10 and Fingerprint sensor is Synaptics 06cb:009a  and machine is Thinkpad T480. I can install and update PAM, but cannot run fprintd-enroll. 

I ran `sudo add-apt-repository ppa:uunicorn/open-fprintd`

And encounter the error: 

`Found existing deb entry in /etc/apt/sources.list.d/uunicorn-ubuntu-open-fprintd-mantic.sources
Hit:1 http://ph.archive.ubuntu.com/ubuntu mantic InRelease
Hit:2 http://security.ubuntu.com/ubuntu mantic-security InRelease              
Hit:3 http://ph.archive.ubuntu.com/ubuntu mantic-updates InRelease             
Hit:4 http://ph.archive.ubuntu.com/ubuntu mantic-backports InRelease           
Hit:5 http://repository.spotify.com stable InRelease                           
Ign:6 https://ppa.launchpadcontent.net/uunicorn/open-fprintd/ubuntu mantic InRelease
Err:7 https://ppa.launchpadcontent.net/uunicorn/open-fprintd/ubuntu mantic Release
  404  Not Found [IP: 185.125.190.80 443]
Reading package lists... Done
E: The repository 'https://ppa.launchpadcontent.net/uunicorn/open-fprintd/ubuntu mantic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
`
The Fingerprint login will show up when i go to **Users** but will disappear once the dialog for **Authentication** opens.

Wambugu-Muchemi, 2024-04-26:
Facing similar issue on Ubuntu 24.04 LTS version. Lenovo thinkpad t480

khatfull, 2024-04-29:
Same as @Wambugu-Muchemi on 24.04 LTS.  ThinkPad x220.  No release file for noble.

khatfull, 2024-04-29:
Can confirm that the workaround in https://github.com/uunicorn/python-validity/issues/188 does NOT work for 24.04.  The installation throws dbus errors.

Wambugu-Muchemi, 2024-04-29:
> Same as @Wambugu-Muchemi on 24.04 LTS.  ThinkPad x220.  No release file for noble.

Oh good to note. Thanks for the heads up, will keep checking

shahzeb-debug, 2024-05-08:
In order to get this work on Ubuntu 24.04 LTS (Noble Numbat) we need a new release file. @uunicorn

shahzeb-debug, 2024-05-08:
Any chance that the developer is active for this project?

earthseekerx, 2024-05-16:
Same problem on a Lenovo ThinkPad T470 running a fresh install of Debian 12.

xkillyridolsx, 2024-05-21:
Similar error trying to get it running on my Thinkpad T480 with Fedora 40. Getting this error:

Copr repo for python-validity owned by tigro                                  1.4 kB/s | 1.5 kB     00:01    
Package fprintd-clients-1:1.94.2-2.fc39.x86_64 is already installed.
Package fprintd-clients-pam-1:1.94.2-2.fc39.x86_64 is already installed.
Error: 
 Problem 1: conflicting requests
  - nothing provides python(abi) = 3.11 needed by open-fprintd-0.6-7.fc39.noarch from copr:copr.fedorainfracloud.org:tigro:python-validity
 Problem 2: conflicting requests
  - nothing provides python(abi) = 3.11 needed by python3-validity-0.14-2.fc39.noarch from copr:copr.fedorainfracloud.org:tigro:python-validity
  - nothing provides python3.11dist(cryptography) >= 2.1.4 needed by python3-validity-0.14-2.fc39.noarch from copr:copr.fedorainfracloud.org:tigro:python-validity
  - nothing provides python3.11dist(pyusb) >= 1 needed by python3-validity-0.14-2.fc39.noarch from copr:copr.fedorainfracloud.org:tigro:python-validity
  - nothing provides python3.11dist(pyyaml) >= 3.12 needed by python3-validity-0.14-2.fc39.noarch from copr:copr.fedorainfracloud.org:tigro:python-validity
(try to add '--skip-broken' to skip uninstallable packages)

shahzeb-debug, 2024-05-21:
> Similar error trying to get it running on my Thinkpad T480 with Fedora 40. Getting this error:
>
> Copr repo for python-validity owned by tigro                                  1.4 kB/s | 1.5 kB [...]

This repository won't work with Fedora 40. This is primarily for Ubuntu I guess. 

You can refer to:

``$ sudo dnf copr enable tigro/python-validity
$ sudo dnf install open-fprintd fprintd-clients fprintd-clients-pam python3-validity
...wait a bit...
$ fprintd-enroll``

Also mentioned in Unicorn's repo. 

However for me this one here: https://copr.fedorainfracloud.org/coprs/taaem/python-validity/

Worked flawlessly on my Thinkpad Yoga 370 with 138a:0097.

alienstro, 2024-05-22:
Same issue ngl

earthseekerx, 2024-05-24:
Any potential workaround for this issue or do we have to wait for the dev to see this issue. Please inform if any steps could be taken to fix this issue with a PR. Good Day.

shahzeb-debug, 2024-05-24:
> Any potential workaround for this issue or do we have to wait for the dev to see this issue. Please inform if any steps could be taken to fix this issue with a PR. Good Day.

I don't think there's any potential solution to the issue unless the dev works on the fix so we have a release file. 

There's a similar repo named taaem's python-validiity that works on Fedora. If someone can port this to Ubuntu then we can get the fingerprint working though I feel this will require a tremendous amount of work. 

https://copr.fedorainfracloud.org/coprs/taaem/python-validity/

shahzeb-debug, 2024-06-03:
This works on Ubuntu 24.04 for 138a:0097 Validity Sensors, Inc.  

Just replace the suites from "noble" to "jammy" and you are all good to go.

[image]

eagle-lk, 2024-06-10:
> This works on Ubuntu 24.04 for 138a:0097 Validity Sensors, Inc.
> 
> Just replace the suites from "noble" to "jammy" and you are all good to go.
> 
> ![Screenshot from 2024-06-03 16-06-44](https://private-user-images.githubusercontent.com/74614482/336038398-aacbc377-1481-4dbe-b446-347a4644571e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc5OTQyMzIsIm5iZiI6MTcxNzk5MzkzMiwicGF0aCI6Ii83NDYxNDQ4Mi8zMzYwMzgzOTgtYWFjYmMzNzctMTQ4MS00ZGJlLWI0NDYtMzQ3YTQ2NDQ1NzFlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjEwVDA0MzIxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZmMThhMGU0MjJhZmNlZTM2ZmQxZjU1ODI0ZDRjNmNiNzY4ZThhOTJmYTU2ODRiNjU1ODA4YmZmMDJmYWYzZTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.3VjoAG_wwuET8qZuenKLpEztqFOX-ZnUO3_zzlzuqMI)

Great！It's work on my ubuntu noble now! Thanks!

eagle-lk, 2024-06-10:
> This works on Ubuntu 24.04 for 138a:0097 Validity Sensors, Inc.
>
> Just replace the suites from "noble" to "jammy" and you are all good to go.
>
> ![Screenshot from 2024-06-03 [...]

Great！It's work on my ubuntu noble now! Thanks!

rasa, 2024-07-31:
`sudo sed -i 's/noble/jammy/' /etc/apt/sources.list.d/uunicorn-ubuntu-open-fprintd-noble.sources`
fixes this on the command line.

vasi, 2024-09-08:
It installed fine, but it seems much less stable. Occasionally my login manager crashes. Maybe we can get a rebuild for noble?

grupofaster, 2024-12-15:
> This works on Ubuntu 24.04 for 138a:0097 Validity Sensors, Inc.
>
> Just replace the suites from "noble" to "jammy" and you are all good to go.
>
> ![Screenshot from 2024-06-03 [...]

Thanks, amazing and simple hack Shahzeb! Regards :rocket: :coffee:

dacog, 2025-03-09:
This is related to #188 . I just posted a summary there in case it helps.