uunicorn/python-validity #192: Python 3.12 breaks compatibility

maxulm, 2024-04-30:
After an upgrade from Python 3.11 to 3.12 python-validity does not longer work:

`Apr 30 10:50:08 limaxpad dbus-service[907]:     return self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
Apr 30 10:50:08 limaxpad dbus-service[907]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 30 10:50:08 limaxpad dbus-service[907]:   File "/usr/lib/python3.12/site-packages/dbus/connection.py", line 634, in call_blocking
Apr 30 10:50:08 limaxpad dbus-service[907]:     reply_message = self.send_message_with_reply_and_block(
Apr 30 10:50:08 limaxpad dbus-service[907]:                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 30 10:50:08 limaxpad dbus-service[907]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: The name does not have an owner
Apr 30 10:50:08 limaxpad dbus-service[907]: During handling of the above exception, another exception occurred:
Apr 30 10:50:08 limaxpad dbus-service[907]: Traceback (most recent call last):
Apr 30 10:50:08 limaxpad dbus-service[907]:   File "/usr/lib/python-validity/dbus-service", line 303, in <module>
Apr 30 10:50:08 limaxpad dbus-service[907]:     main()
Apr 30 10:50:08 limaxpad dbus-service[907]:   File "/usr/lib/python-validity/dbus-service", line 287, in main
Apr 30 10:50:08 limaxpad dbus-service[907]:     bus.get_object('net.reactivated.Fprint', '/net/reactivated/Fprint/Manager')
Apr 30 10:50:08 limaxpad dbus-service[907]:   File "/usr/lib/python3.12/site-packages/dbus/bus.py", line 237, in get_object
Apr 30 10:50:08 limaxpad dbus-service[907]:     return self.ProxyObjectClass(self, bus_name, object_path,
Apr 30 10:50:08 limaxpad dbus-service[907]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 30 10:50:08 limaxpad dbus-service[907]:   File "/usr/lib/python3.12/site-packages/dbus/proxies.py", line 250, in __init__
Apr 30 10:50:08 limaxpad dbus-service[907]:     self._named_service = conn.activate_name_owner(bus_name)
Apr 30 10:50:08 limaxpad dbus-service[907]:                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 30 10:50:08 limaxpad dbus-service[907]:   File "/usr/lib/python3.12/site-packages/dbus/bus.py", line 178, in activate_name_owner
Apr 30 10:50:08 limaxpad dbus-service[907]:     self.start_service_by_name(bus_name)
Apr 30 10:50:08 limaxpad dbus-service[907]:   File "/usr/lib/python3.12/site-packages/dbus/bus.py", line 273, in start_service_by_name
Apr 30 10:50:08 limaxpad dbus-service[907]:     return (True, self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
Apr 30 10:50:08 limaxpad dbus-service[907]:                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 30 10:50:08 limaxpad dbus-service[907]:   File "/usr/lib/python3.12/site-packages/dbus/connection.py", line 634, in call_blocking
Apr 30 10:50:08 limaxpad dbus-service[907]:     reply_message = self.send_message_with_reply_and_block(
Apr 30 10:50:08 limaxpad dbus-service[907]:                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 30 10:50:08 limaxpad dbus-service[907]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: Could not activate remote peer 'net.reactivated.Fprint': startup job failed
`

Filz0r, 2024-05-04:
I had the same issue on a Arch Linux system after the update, but after checking the [wiky](https://wiki.archlinux.org/title/python#Module_not_found_after_Python_version_update), I found that we can manually copy the modules from one place to the other it then starts working, its not ideal but in a pinch it helped me

Qpernicus, 2024-05-04:
> I had the same issue on a Arch Linux system after the update, but after checking the [wiky](https://wiki.archlinux.org/title/python#Module_not_found_after_Python_version_update), I found that we can [...]

Well I tried this and it made it work..I think.. But my SDDM and Hyprlock don't ask for/activate my sensor.. So the service seems active but somewhere the mechanism changed perhaps? I had this which worked:

in /etc/pam.d/hyprlock:

```
auth 	    [success=1 new_authtok_reqd=1 default=ignore]  	pam_unix.so try_first_pass likeauth nullok
auth	    sufficient	pam_fprintd.so
auth        include     login
```

in /etc/pam.d/sddm:

```
auth 	    [success=1 new_authtok_reqd=1 default=ignore]  	pam_unix.so try_first_pass likeauth nullok
auth	    sufficient	pam_fprintd.so
auth        include     system-login
```

Which worked.. but now doesn't .. And i don't have a common-auth file as described in the manual (Also running Arch)

betomoretti, 2024-05-05:
Hi everyone, 
Haven´t tried the workaround yet. But just wanna confirm that I'm having issues with python 3.12.3

maxulm, 2024-05-05:
The last update solved the problem for me

closed 2024-05-05 by maxulm