uunicorn/python-validity #195: verify-no-match after successful enroll [Fedora 40; thinkpad t480s]

coddest, 2024-08-16:
I cannot verify successfully, even after enrolling without issues.

```
[...]
Enroll result: enroll-stage-passed
Enroll result: enroll-completed
coddest@fedora:~$ fprintd-verify
Using device /net/reactivated/Fprint/Device/4
Listing enrolled fingers:
 - #0: right-index-finger
Verify started!
Verifying: any
Verify result: verify-no-match (done)
```

mikpotik, 2024-08-17:
Maybe this will help:

https://github.com/uunicorn/python-validity/issues/189#issuecomment-2294960949

coddest, 2024-08-17:
> Maybe this will help:
> 
> [#189 (comment)](https://github.com/uunicorn/python-validity/issues/189#issuecomment-2294960949)

I had a windows 10 previously installed, but I the fedora I have now setup was clean install, no dual boot.
Do you think it still might be related, would I have to install windows to do the steps you mentioned?

mikpotik, 2024-08-17:
> > Maybe this will help:
> > [#189 (comment)](https://github.com/uunicorn/python-validity/issues/189#issuecomment-2294960949)
> 
> I had a windows 10 previously installed, but the fedora I have now setup was clean install, no dual boot. Do you think it still might be related, would I have to install windows to do the steps you mentioned?

No matter what OS you have or have had installed, I think the registered finger data is stored in the fingerprint recognition module itself. So they will be stored there permanently, regardless of the OS installed. If you used fingerprint recognition on Windows 10, they are in the module. You need to reset them.

mikpotik, 2024-08-17:
Or you can try this, I found it after I solved the problem through WINDOWS 10:

https://github.com/uunicorn/python-validity/issues/167#issuecomment-1931445835
after killed the process that uses your fingerprint sensor. Do a reset according to the author's instructions:

```
$ sudo systemctl stop python3-validity
$ sudo validity-sensors-firmware
$ sudo python3 /usr/share/python-validity/playground/factory-reset.py

# At some of the above points you may get a 'device busy' error,
# depending on how systemctl plays along. Kill offending processes if
# necessary, or re-run the systemctl stop python3-validity command, 
# in case it has automatically been restarted, or or kill other
# offending processes.

$ sudo systemctl start python3-validity
$ fprintd-enroll
```

coddest, 2024-08-18:
I already tried the factory-reset, it did not fix anything.
But I thought I will try again and see how it goes, it still didn't work, but I kept digging around, and broke everything even more, I had to reinstall a lot of things, and this time it worked.

So I have no idea what finally fixed the issue.

closed 2024-08-18 by coddest