uunicorn/python-validity #197: Repository fail to add on Ubuntu 24.04

Decryptus007, 2024-09-06:
W: GPG error: http://repo.mysql.com/apt/ubuntu jammy InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY B7B3B788A8D3785C
E: The repository 'http://repo.mysql.com/apt/ubuntu jammy InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'https://ppa.launchpadcontent.net/uunicorn/open-fprintd/ubuntu noble Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: https://ppa.launchpadcontent.net/git-core/ppa/ubuntu/dists/noble/InRelease: Signature by key E1DD270288B4E6030699E45FA1715D88E1DF1F24 uses weak algorithm (rsa1024)

Tokobotenkai, 2024-09-08:
I saw in the Ubuntu handbook about a workaround here: https://ubuntuhandbook.org/index.php/2024/02/fingerprint-reader-t480s/
There's unofficial PPA for 24.04 here: https://launchpad.net/~ubuntuhandbook1/+archive/ubuntu/open-fprintd

I haven't tried it yet though.

Decryptus007, 2024-09-08:
Oh! Thanks a lot

closed 2024-09-08 by Decryptus007