uunicorn/python-validity #202: Unable to install in Linux Mint 22 Cinnamon - ThinkPad T480 - ID 06cb:009a Synaptics, Inc. Metallica MIS Touch Fingerprint Reader

andremarko, 2024-09-18:
Having issues in adding the repository in Linux Mint 22 Cinnamon:

$ sudo add-apt-repository ppa:uunicorn/open-fprintd
Cannot add PPA: ''This PPA does not support noble''.

andremarko, 2024-09-26:
Thank you for your attention. I will try it as soon as possible and give it a feedback if I managed to solved it!

DSM-Diver, 2024-10-22:
Thanks [bionik](https://github.com/bionik) for the instructions, it worked for me.

Hiabuto-de, 2025-03-28:
### ✅ WORKING Setup for Synaptics Metallica MIS (06cb:009a) on **Linux Mint 22.1 / Ubuntu 24.04 (noble)**

After *hours* of failed attempts, conflicting packages, and endless `facepalm`, I finally got the reader working 🎉

#### 💡 Final Solution (with workaround for noble support):

> Based on @bionik  – just needed a small repo tweak!

---

#### 🛠️ Steps:

```bash
# 1. Add the PPA – but noble isn’t supported, so we edit it after
sudo add-apt-repository ppa:uunicorn/open-fprintd

# 2. Change "noble" to "kinetic"
sudo sed -i 's/noble/kinetic/g' /etc/apt/sources.list.d/uunicorn-open-fprintd.list

# 3. Update + fix conflicting packages
sudo apt update
sudo apt remove fprintd

# 4. Install everything you need
sudo apt install open-fprintd fprintd-clients python3-validity

# 5. Enroll a fingerprint (test if it's working)
fprintd-enroll

# 6. Enable PAM auth
sudo pam-auth-update

# 7. Enable suspend/resume helpers
sudo systemctl enable open-fprintd-resume open-fprintd-suspend
```

---

### ✅ Result:

- Working fingerprint login
- Fully integrated into Linux Mint login & sudo prompts
- Reader: **Synaptics 06cb:009a**
- System: **Linux Mint 22.1 (Xia)** / **Ubuntu 24.04 (noble)** base  
- Kernel: `6.8.x`

> Hope this saves someone from going completely insane 😵‍💫 – shoutout to the whole thread and u/unicorn for the work on this repo ❤️

---

Wenn du magst, kann ich dir noch ne deutsche Version für nen Reddit-Post draus basteln 😸