uunicorn/python-validity #205: "https://ppa.launchpadcontent.net/uunicorn/open-fprintd/ubuntu oracular Release" does not exist, failing the Install.

Nougatflittchen, 2024-11-01:
What i did: sudo add-apt-repository ppa:uunicorn/open-fprintd
What i got: 
Ign:6 https://ppa.launchpadcontent.net/uunicorn/open-fprintd/ubuntu oracular InRelease
Err:7 https://ppa.launchpadcontent.net/uunicorn/open-fprintd/ubuntu oracular Release
  404  Not Found [IP: 2620:2d:4000:1::81 443]
Reading package lists... Done
E: The repository 'https://ppa.launchpadcontent.net/uunicorn/open-fprintd/ubuntu oracular Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

Very new to linux, i can't find a fix for this online....

dacog, 2025-03-09:
check #188 

In your case you are using [Ubuntu 24.10](https://releases.ubuntu.com/oracular) Oracular Oriole. The workaround in issue #188 is confirmed to work up to Ubuntu Noble 24.04 (one version before). You can try using the repo for `jammy`  and update in that issue if it worked or not.

uunicorn, 2025-06-10:
I just published PPA packages for Noble. I can copy binaries for Oracular if someone can confirm Noble packages are working.