uunicorn/python-validity #206: `FileNotFoundError: [Errno 2] No such file or directory` with `06cb:009a Synaptics, Inc. Metallica MIS Touch Fingerprint Reader`

Inglan, 2024-11-08:
In the archwiki page of my laptop, it recommends this project for fingerprint reader (https://wiki.archlinux.org/title/Lenovo_ThinkPad_X1_Carbon_(Gen_6)#Fingerprint_reader).

After installing from the AUR, and trying to enroll, it shows the `list_devices failed:` error. When I view logs, I can see there is a `No such file or directory` error in the systemd logs. I have a `06cb:009a Synaptics, Inc. Metallica MIS Touch Fingerprint Reader`  on an X1 Carbon Gen 6

Logs:
```
→ fprintd-enroll
list_devices failed:
→ sudo systemctl status python3-validity
[sudo] password for ingo:
× python3-validity.service - python-validity driver dbus service
     Loaded: loaded (/usr/lib/systemd/system/python3-validity.service; disabled; preset: disabled)
     Active: failed (Result: exit-code) since Fri 2024-11-08 12:00:52 AEDT; 5min ago
   Duration: 1.034s
 Invocation: 40dc7d167dac462c9027fcf6527a0809
    Process: 1036187 ExecStart=/usr/lib/python-validity/dbus-service --debug (code=exited, status=1/FAILURE)
   Main PID: 1036187 (code=exited, status=1/FAILURE)
   Mem peak: 27.8M
        CPU: 488ms

Nov 08 12:00:52 smasbook-pro-max dbus-service[1036187]:   File "/usr/lib/python3.12/site-packages/validitysensor/init.py", line 50, in open
Nov 08 12:00:52 smasbook-pro-max dbus-service[1036187]:     open_common()
Nov 08 12:00:52 smasbook-pro-max dbus-service[1036187]:   File "/usr/lib/python3.12/site-packages/validitysensor/init.py", line 35, in open_common
Nov 08 12:00:52 smasbook-pro-max dbus-service[1036187]:     upload_fwext()
Nov 08 12:00:52 smasbook-pro-max dbus-service[1036187]:   File "/usr/lib/python3.12/site-packages/validitysensor/upload_fwext.py", line 49, in upload_fwext
Nov 08 12:00:52 smasbook-pro-max dbus-service[1036187]:     with open(fw_path, 'rb') as f:
Nov 08 12:00:52 smasbook-pro-max dbus-service[1036187]:          ^^^^^^^^^^^^^^^^^^^
Nov 08 12:00:52 smasbook-pro-max dbus-service[1036187]: FileNotFoundError: [Errno 2] No such file or directory: '/var/run/python-validity//6_07f_lenovo_mis_qm.xpfwext'
Nov 08 12:00:52 smasbook-pro-max systemd[1]: python3-validity.service: Main process exited, code=exited, status=1/FAILURE
Nov 08 12:00:52 smasbook-pro-max systemd[1]: python3-validity.service: Failed with result 'exit-code'.
→ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
[REMOVED]
Bus 001 Device 014: ID 06cb:009a Synaptics, Inc. Metallica MIS Touch Fingerprint Reader
[REMOVED]
```

danknil, 2024-11-29:
i had exact same issue with exact same laptop. Fixed by going through https://github.com/uunicorn/python-validity?tab=readme-ov-file#errors-on-startup

Inglan, 2024-12-04:
Randomly started working... Think it wasn't working because I was using linux-zen

closed 2024-12-04 by Inglan