uunicorn/python-validity #213: python3-validity Service Fails Due to Exception 'Refusing to start more than 10 times per minute'

7imeout, 2025-04-18:
About a month ago, I lost the fingerprint auth functionality after running some system updates on my 6th-gen ThinkPad X1 Carbon running Ubuntun 24.04 LTS Desktop. 

I'm not sure which change from the update may have been the root cause here, but: `sudo systemctl status python3-validity` reports service failure due to the exception raised from `dbus-service`, line 217:
```
    if len(lines) > 10:
        raise Exception('Refusing to start more than 10 times per minute')
```

I discovered today that simply commenting out this exception keeps the service running just fine (AFAICT). I'm not exactly sure what the consequences are for letting the service start more than 10 times per minute, but based on the comment, it seem this was some sort of a workaround to get `systemd` to back off?

If the dev or anyone who knows better could enlighten me, I'd be grateful. Also happy to look into what this `systemd` issue is and how one might get it to "properly" back off. But I need a bit more context as to what that means. I.e., what are we trying to get `systemd` to back off from?

`lsb_release -a` output:
```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.2 LTS
Release:	24.04
Codename:	noble
```

pek-trio, 2025-05-09:
Hello @7imeout , I am facing the same issue. Could you point me to the exact file where you commented that exception? I would check if it helps me too :)
Thanks in advance!

7imeout, 2025-05-23:
@pek-trio Sorry for the late response! The stack trace you see should have the path. For me, it was:
```
/usr/lib/python-validity/dbus-service
```