uunicorn/python-validity #214: Question regards a specific supported fingerprint sensor

TamFishe, 2025-04-22:
Hi, I have a HP EliteBook 2560p with a palmrest from a 2570p that has the Validity Sensor VFS491. Is it supported under these drivers using Debian 12? I would like to make the laptop under Debian 12 with KDE Plasma and I wonder if fingerprint functionality can be present before I nuke my Windows 10 installation

gabrielttavares, 2025-04-22:
I don't think so, unless KDE fixes it

TamFishe, 2025-04-22:
I see, thanks for the reply though. For a couple of hours I've been digging the depths of the internet pretty much and i think probably the driver that makes the VFS495 to work might make the 491 to work aswell. It's a pain though having to find a good driver for an old laptop which mostly people probably moved on or use with windows and plan to ditch it after w10 reaches EoL