uunicorn/python-validity #231: T480 (06cb:009a) - python3-validity runs, but open-fprintd finds no devices

eduardo-pcmr, 2025-09-13:
Hello,

I'm having an issue with my ThinkPad T480 fingerprint reader (USB ID 06cb:009a) on Arch Linux.

The python-validity-git package fails to start with a Signature validation failed: 0x4f04 error.

After a lot of troubleshooting (including a cold boot and trying helper scripts), I was able to get the service running by manually building the stable python-validity package from the AUR.

Now, systemctl status python3-validity.service shows active (running) and the logs are clean.

However, fprintd-enroll still fails with list_devices failed:. The log for open-fprintd.service shows DEBUG:root:no devices. This happens even after restarting both python3-validity.service and open-fprintd.service in the correct order.

It seems the driver service is running correctly but fails to advertise the device to open-fprintd via D-Bus.

Log output for python3-validity.service:
[~]$ systemctl status python3-validity.service
● python3-validity.service - python-validity driver dbus service
     Loaded: loaded (/usr/lib/systemd/system/python3-validity.service; enabled;>
     Active: active (running) since Fri 2025-09-12 21:46:07 -03; 192ms ago
 Invocation: 42e51b9086d54393a4d365942c6c2e93
   Main PID: 2158 (python3)
      Tasks: 2 (limit: 18879)
     Memory: 20.6M (peak: 20.6M)
        CPU: 173ms
     CGroup: /system.slice/python3-validity.service
             └─2158 python3 /usr/lib/python-validity/dbus-service --debug

set 12 21:46:07 t480 systemd[1]: python3-validity.service: Scheduled restart jo>
set 12 21:46:07 t480 systemd[1]: Started python-validity driver dbus service.
set 12 21:46:07 t480 python3[2158]: DEBUG:root:>cmd> 3e
[~]$

Log output for open-fprintd.service:
[~]$ systemctl status open-fprintd.service
● open-fprintd.service - Open FPrint Daemon
     Loaded: loaded (/usr/lib/systemd/system/open-fprintd.service; static)
     Active: active (running) since Fri 2025-09-12 21:38:16 -03; 9min ago
 Invocation: 7ff94d949e60482e81ef8c3525dcc272
   Main PID: 1653 (open-fprintd)
      Tasks: 1 (limit: 18879)
     Memory: 13.9M (peak: 14.2M)
        CPU: 104ms
     CGroup: /system.slice/open-fprintd.service
             └─1653 /usr/bin/python3 /usr/lib/open-fprintd/open-fprintd --debug

set 12 21:38:15 t480 systemd[1]: Starting Open FPrint Daemon...
set 12 21:38:16 t480 systemd[1]: Started Open FPrint Daemon.
set 12 21:38:16 t480 open-fprintd[1653]: /usr/lib/open-fprintd/open-fprintd:25:>
set 12 21:38:16 t480 open-fprintd[1653]:   GObject.MainLoop().run()
set 12 21:38:42 t480 open-fprintd[1653]: DEBUG:root:GetDefaultDevice
set 12 21:38:42 t480 open-fprintd[1653]: DEBUG:root:no devices
[~]$ 

Thank you for your work on this project.

compicat, 2025-09-13:
i m experiencing same problem lately. is there any solution? device is same t480.

newv-cell, 2025-09-14:
same problem, i guess there is no fix